tinygo-org/tinygo #2493: no usable wasm-opt found, update or run "make binaryen"

patrickhuber, 2022-01-06:
I'm running on windows 11 Version 10.0.22000 Build 22000 and I received an error 'no usable wasm-opt found, update or run "make binaryen"' while trying to compile to wasm. 

I installed tinygo using scoop per the instructions [here](https://tinygo.org/getting-started/install/windows/) and I'm compiling a simple main.go:

```
package main

import "fmt"

func main() {
    fmt.Println("Hello World!")
}
```

```
 > tinygo build -o wasm.wasm -target wasm main.go
 no usable wasm-opt found, update or run "make binaryen"
        C:\Users\patri\scoop\apps\tinygo\current/bin/wasm-opt.exe: exit status 0xc0000135
```

here is the output of go env

```
> go env
set GO111MODULE=
set GOARCH=amd64
set GOBIN=
set GOCACHE=C:\Users\patri\AppData\Local\go-build
set GOENV=C:\Users\patri\AppData\Roaming\go\env
set GOEXE=.exe
set GOEXPERIMENT=
set GOFLAGS=
set GOHOSTARCH=amd64
set GOHOSTOS=windows
set GOINSECURE=
set GOMODCACHE=C:\Users\patri\go\pkg\mod
set GONOPROXY=
set GONOSUMDB=
set GOOS=windows
set GOPATH=C:\Users\patri\go
set GOPRIVATE=
set GOPROXY=https://proxy.golang.org,direct
set GOROOT=C:\Program Files\Go
set GOSUMDB=sum.golang.org
set GOTMPDIR=
set GOTOOLDIR=C:\Program Files\Go\pkg\tool\windows_amd64
set GOVCS=
set GOVERSION=go1.17.5
set GCCGO=gccgo
set AR=ar
set CC=gcc
set CXX=g++
set CGO_ENABLED=1
set GOMOD=NUL
set CGO_CFLAGS=-g -O2
set CGO_CPPFLAGS=
set CGO_CXXFLAGS=-g -O2
set CGO_FFLAGS=-g -O2
set CGO_LDFLAGS=-g -O2
set PKG_CONFIG=pkg-config
set GOGCCFLAGS=-m64 -mthreads -fno-caret-diagnostics -Qunused-arguments -fmessage-length=0 -fdebug-prefix-map=C:\Users\patri\AppData\Local\Temp\go-build1753883593=/tmp/go-build -gno-record-gcc-switches
```

here is the output of tinygoenv

```
> tinygo env
GOOS="windows"
GOARCH="amd64"
GOROOT="C:\\Program Files\\Go"
GOPATH="C:\\Users\\patri\\go"
GOCACHE="C:\\Users\\patri\\AppData\\Local\\tinygo"
CGO_ENABLED="1"
TINYGOROOT="C:\\Users\\patri\\scoop\\apps\\tinygo\\current"
```

niaow, 2022-01-11:
It looks like there is a missing DLL somehow. I am not entirely sure what it could be since I tried to statically link it. My guess is that the linker ignored the request to statically link mingw's compiler runtime, but I will need to check that next time I boot up Windows.

niaow, 2022-01-11:
It might make sense to swap over to scoop's copy of binaryen rather than attempting to figure out what happened here. It is new enough now.

deadprogram, 2022-01-12:
> It might make sense to swap over to scoop's copy of binaryen rather than attempting to figure out what happened here. It is new enough now.

Can we do that for all platforms (Windows, macOS, Linux) now?

niaow, 2022-01-12:
I don't think we really can on Linux, but for the others I think so

niaow, 2022-01-15:
I am working on the switch over in #2527

niaow, 2022-01-15:
@patrickhuber We should be able to fix this for the next release. However if you need a workaround for now, you can `scoop install binaryen`.

closed 2022-01-26 by deadprogram