tinygo-org/tinygo #2494: build failed on macos.

clearcodecn, 2022-01-07:
```
☁  rasproxy  tinygo version
tinygo version 0.21.0 darwin/amd64 (using go version go1.16 and LLVM version 11.0.0)
☁  rasproxy  go version
go version go1.16 darwin/amd64
☁  rasproxy  go env
GO111MODULE=""
GOARCH="amd64"
GOBIN=""
GOCACHE="/Users/zhangmengjin/Library/Caches/go-build"
GOENV="/Users/zhangmengjin/Library/Application Support/go/env"
GOEXE=""
GOFLAGS=""
GOHOSTARCH="amd64"
GOHOSTOS="darwin"
GOINSECURE=""
GOMODCACHE="/Users/zhangmengjin/go/pkg/mod"
GONOPROXY="git.dustess.com,go.clearcode.cn"
GONOSUMDB="git.dustess.com,go.clearcode.cn"
GOOS="darwin"
GOPATH="/Users/zhangmengjin/go"
GOPRIVATE="git.dustess.com,go.clearcode.cn"
GOPROXY="https://goproxy.io,https://goproxy.cn,direct"
GOROOT="/Users/zhangmengjin/.g/go"
GOSUMDB="sum.golang.org"
GOTMPDIR=""
GOTOOLDIR="/Users/zhangmengjin/.g/go/pkg/tool/darwin_amd64"
GOVCS=""
GOVERSION="go1.16"
GCCGO="gccgo"
AR="ar"
CC="clang"
CXX="clang++"
CGO_ENABLED="1"
GOMOD="/Users/zhangmengjin/Desktop/projects/rasproxy/go.mod"
CGO_CFLAGS="-g -O2"
CGO_CPPFLAGS=""
CGO_CXXFLAGS="-g -O2"
CGO_FFLAGS="-g -O2"
CGO_LDFLAGS="-g -O2"
PKG_CONFIG="pkg-config"
GOGCCFLAGS="-fPIC -arch x86_64 -m64 -pthread -fno-caret-diagnostics -Qunused-arguments -fmessage-length=0 -fdebug-prefix-map=/var/folders/wp/1gb_nj116jx50p8qh38tqpbw0000gn/T/go-build39270924=/tmp/go-build -gno-record-gcc-switches -fno-common"
☁  rasproxy

```

```
☁  rasproxy tinygo build -o bin/rasproxy cmd/main.go
# text/template
/Users/zhangmengjin/.g/go/src/text/template/exec.go:376:17: val.Type().ChanDir undefined (type reflect.Type has no field or method ChanDir)
/Users/zhangmengjin/.g/go/src/text/template/exec.go:376:38: SendDir not declared by package reflect
/Users/zhangmengjin/.g/go/src/text/template/exec.go:382:20: val.Recv undefined (type reflect.Value has no field or method Recv)
/Users/zhangmengjin/.g/go/src/text/template/exec.go:608:19: ptr.MethodByName undefined (type reflect.Value has no field or method MethodByName)
/Users/zhangmengjin/.g/go/src/text/template/exec.go:615:33: receiver.Type().FieldByName undefined (type reflect.Type has no field or method FieldByName)
/Users/zhangmengjin/.g/go/src/text/template/exec.go:650:21: etyp.FieldByName undefined (type reflect.Type has no field or method FieldByName)
/Users/zhangmengjin/.g/go/src/text/template/exec.go:683:9: typ.IsVariadic undefined (type reflect.Type has no field or method IsVariadic)
/Users/zhangmengjin/.g/go/src/text/template/exec.go:684:18: typ.NumIn undefined (type reflect.Type has no field or method NumIn)
/Users/zhangmengjin/.g/go/src/text/template/exec.go:686:79: typ.NumIn undefined (type reflect.Type has no field or method NumIn)
/Users/zhangmengjin/.g/go/src/text/template/exec.go:688:25: typ.NumIn undefined (type reflect.Type has no field or method NumIn)
/Users/zhangmengjin/.g/go/src/text/template/exec.go:689:69: typ.NumIn undefined (type reflect.Type has no field or method NumIn)
/Users/zhangmengjin/.g/go/src/text/template/exec.go:693:71: typ.NumOut undefined (type reflect.Type has no field or method NumOut)
/Users/zhangmengjin/.g/go/src/text/template/exec.go:700:32: typ.In undefined (type reflect.Type has no field or method In)
/Users/zhangmengjin/.g/go/src/text/template/exec.go:703:9: typ.IsVariadic undefined (type reflect.Type has no field or method IsVariadic)
/Users/zhangmengjin/.g/go/src/text/template/exec.go:704:18: typ.In undefined (type reflect.Type has no field or method In)
/Users/zhangmengjin/.g/go/src/text/template/exec.go:704:25: typ.NumIn undefined (type reflect.Type has no field or method NumIn)
/Users/zhangmengjin/.g/go/src/text/template/exec.go:711:12: typ.In undefined (type reflect.Type has no field or method In)
/Users/zhangmengjin/.g/go/src/text/template/exec.go:711:19: typ.NumIn undefined (type reflect.Type has no field or method NumIn)
/Users/zhangmengjin/.g/go/src/text/template/exec.go:712:10: typ.IsVariadic undefined (type reflect.Type has no field or method IsVariadic)
/Users/zhangmengjin/.g/go/src/text/template/exec.go:716:13: typ.In undefined (type reflect.Type has no field or method In)
/Users/zhangmengjin/.g/go/src/text/template/funcs.go:95:88: v.Type().NumOut undefined (type reflect.Type has no field or method NumOut)
/Users/zhangmengjin/.g/go/src/text/template/funcs.go:113:11: typ.NumOut undefined (type reflect.Type has no field or method NumOut)
/Users/zhangmengjin/.g/go/src/text/template/funcs.go:115:11: typ.NumOut undefined (type reflect.Type has no field or method NumOut)
/Users/zhangmengjin/.g/go/src/text/template/funcs.go:115:32: typ.Out undefined (type reflect.Type has no field or method Out)
/Users/zhangmengjin/.g/go/src/text/template/funcs.go:290:14: item.Slice3 undefined (type reflect.Value has no field or method Slice3)
/Users/zhangmengjin/.g/go/src/text/template/funcs.go:322:92: typ.NumOut undefined (type reflect.Type has no field or method NumOut)
/Users/zhangmengjin/.g/go/src/text/template/funcs.go:324:15: typ.NumIn undefined (type reflect.Type has no field or method NumIn)
/Users/zhangmengjin/.g/go/src/text/template/funcs.go:326:9: typ.IsVariadic undefined (type reflect.Type has no field or method IsVariadic)
/Users/zhangmengjin/.g/go/src/text/template/funcs.go:330:17: typ.In undefined (type reflect.Type has no field or method In)
/Users/zhangmengjin/.g/go/src/text/template/funcs.go:341:11: typ.IsVariadic undefined (type reflect.Type has no field or method IsVariadic)
/Users/zhangmengjin/.g/go/src/text/template/funcs.go:342:18: typ.In undefined (type reflect.Type has no field or method In)
make: *** [tiny-build] Error 1
☁  rasproxy

```

mar1n3r0, 2022-01-09:
> ```
> ☁  rasproxy  tinygo version
> tinygo version 0.21.0 darwin/amd64 (using go version go1.16 and LLVM version 11.0.0)
> ☁  rasproxy  go version
> go version go1.16 darwin/amd64
> ☁  rasproxy  go env [...]

https://tinygo.org/docs/reference/lang-support/stdlib/#texttemplate

aykevl, 2022-01-21:
We don't fully support reflect yet. The reflect package is needed for the text/template package.

sallaben, 2022-07-05:
@aykevl What can I do to implement reflect.MakeMap and reflect.MakeMapWithSize, is that feasible or is there a reason we can't?

aykevl, 2022-07-15:
@sallaben I think it's possible at the moment. What you need to do is essentially replicate `createMakeMap` in the reflect package:

https://github.com/tinygo-org/tinygo/blob/b65447c7d567eea495805656f45472cc3c483e03/compiler/map.go#L20-L58

sallaben, 2022-07-15:
Very good to know. Thanks for the insight! I’ll give it a shot.

mymmrac, 2024-03-06:
Hello, @aykevl are there any updates/plans/roadmap to support more reflection? 

With v0.31.1 fails in runtime because of reflection

aykevl, 2024-03-16:
I'm not aware of anybody working on this at the moment.

dgryski, 2024-10-03:
`text/template` now builds but the tests fail to pass because we don't have support for reflection for functions.

elewis787, 2024-10-24:
> `text/template` now builds but the tests fail to pass because we don't have support for reflection for functions.

@dgryski - are there any plans to support `(reflect.Value).MethodByName()` or other required text/templating refection capabilities?  

I've gone through the reflection issues I was able to find - any additional resources/open prs I can read through to understand this more?

elewis787, 2024-10-31:
I have started some very basic work on supporting functions. Not sure what problem will come from it but most of the func signature data seems to be available.  So far, NumIn(), NumOut(), and IsVariadric() have been implemented, but there is a lot left to test the implementations. 

Next on the list in In() and Out(). Eventually, I'll take a look at Method and MethodByName but haven't gotten too far. 

If any hidden bombs could prevent this that folks know about please let me know! Otherwise, I'll keep pushing forward until I hit one 😎

elewis787, 2024-10-31:
some progress can be found in my branch https://github.com/elewis787/tinygo/tree/elewis-function-reflect 

working with @jeff1010322 to see if we can get tests running for the reflect package. 

I will still need `MethodByName` and potentially a few others for the text/template package to be usable.