tinygo-org/tinygo #2496: Enhancement request: inline WAT

jzabinski-dolios, 2022-01-07:
I have various functions written and optimized in WAT. This is sometimes important to maximize performance. Being able to inline these functions similar to the ability to [inline assembly](https://tinygo.org/docs/concepts/compiler-internals/inline-assembly/) would be handy.

aykevl, 2022-01-19:
We do support inline assembly, but it is generally only used for microcontrollers. It might actually work for WebAssembly. See https://tinygo.org/docs/concepts/compiler-internals/inline-assembly/ for details, but use `import "device"` instead of `import "device/arm"`.

The inline assembly for WebAssembly would not be WAT, but the other text format used internally in compilers. This is what it looks like: https://github.com/tinygo-org/tinygo/blob/release/src/internal/task/task_asyncify_wasm.S

jzabinski-dolios, 2022-01-19:
Interesting possibility. Supporting both microcontrollers and WASM with one code set would be useful.

Is there an official name for the text format used by the compiler? (I want to educate myself on the text format.)

aykevl, 2022-01-21:
> Is there an official name for the text format used by the compiler? (I want to educate myself on the text format.)

Not sure, I think it's just called "assembly" (which is a bit of an overloaded term, I know).

codefromthecrypt, 2022-09-07:
I would recommend against using WAT format inside tinygo, not because it isn't neat, but because it conflicts with who owns the module. For example, the result of compilation is one module, yet each `%.wat` file would be its own module by definition. That makes some strange merge concerns even if TinyGo wanted to include a wat parser in its codebase.

Is it ok to close this? I suspect if there is interest later, provided the issue isn't locked, someone can still scream.

jzabinski-dolios, 2022-09-07:
Reconsidering the issue, having inline WAT would result in superior WASM, but I suspect that the same result could be achieved with some compiler gymnastics: converting the optimized WAT to WASM using WABT, WASM to C again [using WABT](https://github.com/WebAssembly/wabt/blob/main/wasm2c/README.md), and C to IR using what-have-you. Then you could compile the TinyGo to IR separately, and compile both code sets to WASM.

Not that I have tried yet: that's firmly in the 'future projects' category for us. FIne to close for now.

closed 2022-09-07 by jzabinski-dolios

codefromthecrypt, 2022-09-08:
thanks for the thoughts. I think it is reasonable and interesting to ask about, just wanted to help you from holding breath on it :D

If you see another compiler doing this successfully, ping back notes! I would suspect ones that have larger communities working on wasm, such as rust or emscripten, to be able to do this before we would. At the moment, they seem to be following the same approach, which is to allow embedding an assembly file

https://github.com/emscripten-core/emscripten/pull/15913/files#diff-7672a23929e7d74baa102dc6b482527e85c45259f8d01f6afddf58a32f68afe8

codefromthecrypt, 2022-09-08:
also another option that is maybe not what you are looking for, but what we do in wazero. have multiple modules and your code import the fast-pathed ones compiled to wasm. ack this gets back into integration, but wasm-wasm calls should be fast even if cross module boundaries. We do this in wazero in order to stub out unused callbacks far faster than native functions.