tinygo-org/tinygo #2507: potential regression with tinygo 0.21 with regards to 0.20

stephane-moreau, 2022-01-10:
I've built a domain specific expression evaluation in go and I use tinygo to generate a WASM that exposes expose it wihtin the browser.
I did notice a panic with a stack fault in the browser nce code was compiled with tinygo 0.21.0

Code is working fine when WASM is generated by tinygo 0.20.0
tinygo version 0.20.0 linux/amd64 (using go version go1.17.6 and LLVM version 11.0.0)

when updating to tinygo 0.21.0 
tinygo version 0.21.0 linux/amd64 (using go version go1.17.6 and LLVM version 11.0.0)

here is the output I got from tests run using version 0.21.0
```bash
panic: runtime error: stack overflow
syscall/js.finalizeRef not implemented
panic: JavaScript error: unreachable
RuntimeError: unreachable
    at runtime._panic (wasm://wasm/00960fce:wasm-function[29]:0x1f9b)
    at (syscall/js.Value).Call (wasm://wasm/00960fce:wasm-function[296]:0x27348)
    at imagino/wasm/expr.TestConstantComplexExprCompilation (wasm://wasm/00960fce:wasm-function[1004]:0xdde69)
    at imagino/wasm/expr.TestMain (wasm://wasm/00960fce:wasm-function[1006]:0xe0283)
    at main.main (wasm://wasm/00960fce:wasm-function[262]:0x217d3)
    at runtime.run$1 (wasm://wasm/00960fce:wasm-function[260]:0x2126b)
    at runtime.run$1$gowrapper (wasm://wasm/00960fce:wasm-function[258]:0x21117)
    at tinygo_launch (wasm://wasm/00960fce:wasm-function[1008]:0xe085b)
    at (*internal/task.Task).Resume (wasm://wasm/00960fce:wasm-function[141]:0xec63)
    at runtime.scheduler (wasm://wasm/00960fce:wasm-function[259]:0x211d8)
FAIL    imagino/wasm/expr       1.089s
FAIL
```

I'm still not able to simplify the code enough to provide a  standalone code for replication.
Is there any way I can get more insight on why I'm getting the stack overflow

deadprogram, 2022-01-10:
Hello @stephane-moreau is there any way that you can try this out with the latest `dev` branch?

stephane-moreau, 2022-01-10:
I'll check this within the next few days building tinygo from sources

niaow, 2022-01-11:
The stack overflow part is most likely fixed by #2447 on dev.

stephane-moreau, 2022-02-01:
Unfortunately did not had time to check this issue before release 0.22 
but I can now confirm that  it's fixed in release 0.22 with version 105 of binaryen (thanks @niaow and @deadprogram )

Closing issue

closed 2022-02-01 by stephane-moreau