tinygo-org/tinygo #2509: Break up main.go

cweagans, 2022-01-10:
I recently opened a PR to add JSON output to `tinygo info` and I was surprised to see how big main.go was. If I took the time to break main.go up into smaller parts and perhaps re-build it around something like mitchellh/cli or urfave/cli , would there be some interest in merging that? A 2k line file is somewhat unwieldy :)

dkegel-fastly, 2022-01-11:
I would advise caution there.  Please avoid importing new dependencies.

For what it's worth, tinygo's source tree is generally patterned after upstream go's source
tree, and sometimes it makes sense to use the same filenames.
Upstream puts main in src/cmd/go/main.go.
I could imagine tinygo's main.go moving to src/cmd/tinygo/main.go at some point,
and becoming a bit more similar to it.

But I mostly just fiddle around in the standard library, and I'm a relative newcomer to the project, so that that with a grain of salt.

[I would suggest making titles of issues reflect the problem rather than the proposed solution; that helps make sure other fixes are considered.]

aykevl, 2022-01-21:
We want the command line interface to stay similar to the `go` toolchain, for compatibility.
I'm not strongly against new dependencies (if they aren't too complicated) but also don't see a big benefit in splitting main.go up. I don't think the source code will become much more readable because then you have to look into multiple files. But that's just my opinion.

deadprogram, 2024-12-04:
Closing since no action required and question was answered. Thanks everyone!