tinygo-org/tinygo #2512: WASM: strange behavior when multi-value is used

jzabinski-dolios, 2022-01-11:
dev branch

Go [supports](https://go.dev/tour/basics/6) the use of multiple results. [So does WebAssembly](https://webassembly.org/roadmap/).

If I write a simple Go function that returns multiple values:
```
//export multiValue
func multiValue() (int, int) {
	return 1, 2
}
```

Ideally that would result in the WASM equivalent of this WAT:
```
(func $[number] (result i32 i32)
  i32.const 1
  i32.const 2
)
```

Instead the following is generated:
```
(func $34 (param $0 i32)
  (call $18
   (local.get $0)
  )
  (call $21)
 )

(func $18 (param $0 i32)
  (i64.store
   (local.get $0)
   (i64.const 8589934593)
  )
 )

(func $20
 )

 (func $21
  (call $20)
  (call $20)
 )
```

Basically it seems to:

1. Create a parameter that expects an i32
2. Stores the value from 1 somewhere in linear memory
3. Calls a function twice that does nothing

My guess is that TinyGo does not currently support multiple results, but the [documentation](https://tinygo.org/docs/reference/lang-support/) implies to me that it is supported. (It says that 'all basic types and all regular control flow' is supported. I think of multiple function results as a basic variable type.)

niaow, 2022-01-11:
While multi-value returns are supported by WASM, they are not supported by the WASM C ABI. They are represented as a struct. As for the parameter, what is happening there is that the WASM C ABI passes the return value indirectly.

In other words, the underlying function type is effectively `func() struct{ x, y int }`, which gets implemented by the WASM C ABI as `func(dst *struct{ x, y int })`.

As for the `.command_export` I am not entirely sure, but it seems to be from https://reviews.llvm.org/D81689.

References:
- https://github.com/WebAssembly/tool-conventions/blob/main/BasicCABI.md

niaow, 2022-01-12:
It does return 2 values through the pointer, but it does so with a single store. If you take the i32 values and concatenate them (`(2*(2^32))+1`) you get the argument to the 64-bit store (`8589934593`).

jzabinski-dolios, 2022-01-12:
Oh thank you, I missed that!

Now that we have a little more information, I was able to get this working fine. For posterity:

If you have a Go function like this:

```
//export multiValue
func multiValue() (int, int) {
	return 1, 2
}
```

And we use these settings for `tinygo build`:
```
-no-debug
-target=wasm
-gc=none
-scheduler=none
-panic=trap
```

TinyGo will compile that to the equivalent WASM of the following WAT function:
```
(func $[some number] (param $0 i32)
  (i64.store
   (local.get $0)
   (i64.const 8589934593)
  )
 )
```

Basically the function is looking for an address in its linear memory in which to put the function result. It is expecting the embedder to know three things:

1. The size of the bytes to be stored by the function called
2. The nature of the stored bytes (int, float64, etc)
2. A safe place within linear memory to put its bytes

And if you want to use those values from the embedding Javascript, all you need to do is access the embedded linear memory. (Made accessible elsewhere in the WAT/WASM via `(export "memory" (memory $0))`.)

The way to call `multiValue()` within Go would be `a,b := multiValue()`.

The way to call `multiValue()` within (NodeJS) Javascript would be:
```
const fs = require('fs');

const memStorageBytes = fs.readFileSync('./[some file].wasm');

WebAssembly.instantiate(new Uint8Array(memStorageBytes))
    .then(obj => obj.instance.exports)
    .then(exported => {
        // Tell multiValue to do its thing, storing results beginning at byte offset 0
        exported.multiValue(0);
        const results = new Uint32Array(exported.memory.buffer);
        const [a,b] = results;
        console.log(a, b);
    });
```

Result from the console should be `1 2`.

closed 2022-01-12 by jzabinski-dolios

dgryski, 2022-01-13:
This seems like the sort of thing that ought to be documented somewhere obvious for future reference.

aykevl, 2022-01-21:
The way that Go code looks and the way it is represented in WebAssembly are two very different things. They happen to match in many cases, but in many other cases they don't. For example, Go (and TinyGo) support `int64`, but older browsers do not support `i64` in the WebAssembly<->JS boundary. Therefore it is emitted as a pointer to an object. Another example: Go has `int8`, but WebAssembly hasn't. So it is extended to `i32`.

We could document this. If we do that, this would be the most appropriate place: https://tinygo.org/docs/concepts/compiler-internals/calling-convention/