tinygo-org/tinygo #2516: avr: interface lowering problem

deadprogram, 2022-01-12:
The CI for the TinyGo drivers repo just failed on one of the smoke tests seemingly due to some recent change in the last week:

https://app.circleci.com/pipelines/github/tinygo-org/drivers/1097/workflows/da7ef58e-3992-46be-876b-e085efe46756/jobs/1132?invite=true#step-105-362

```
tinygo build -size short -o ./build/test.hex -target=arduino ./examples/keypad4x4/main.go
panic: expected inttoptr

goroutine 82 [running]:
github.com/tinygo-org/tinygo/transform.LowerFuncValues({0x7f4f9849cc40})
	/tinygo/transform/func-lowering.go:186 +0xbb8
github.com/tinygo-org/tinygo/transform.Optimize({0xc0000a5de0}, 0xc0002ca4b0, 0x2, 0x0, 0x5)
	/tinygo/transform/optimizer.go:83 +0x4b8
github.com/tinygo-org/tinygo/builder.optimizeProgram({0x53469a0}, 0xc0002ca4b0)
	/tinygo/builder/build.go:855 +0x1f4
github.com/tinygo-org/tinygo/builder.Build.func2(0x0)
	/tinygo/builder/build.go:473 +0x66d
github.com/tinygo-org/tinygo/builder.runJob(0xc000207440, 0x0)
	/tinygo/builder/jobs.go:222 +0x4f
created by github.com/tinygo-org/tinygo/builder.runJobs
	/tinygo/builder/jobs.go:123 +0x5f8
make: *** [Makefile:190: smoke-test] Error 2
```

The last time this test ran successfully was about 7 days ago. :crying_cat_face:

dkegel-fastly, 2022-01-18:
I just ran into this here on a different app.

dkegel-fastly, 2022-01-18:
Bisecting (using 'go install' as the build, with llvm11 from brew) finds the first bad commit is
```
commit ebd4969cde351cdc31eb142f14dd4c1ce1352de1 (refs/bisect/bad)
Author: Ayke van Laethem <aykevanlaethem@gmail.com>
Date:   Thu Dec 9 02:52:08 2021 +0100

    all: switch to LLVM 13
```

dkegel-fastly, 2022-01-18:
However, the problem occurs even if built using "make" and the bundled llvm13, I think.

dkegel-fastly, 2022-01-18:
When I tried rewinding to the previous commit, and using the bundled llvm 12, I got a bit more output before the error.
```
compiling temp package: compile tinygo: '/Users/dkegel/src/tinygo/src/internal/task/task_asyncify_wasm.S:13:5: error: symbol stop_rewind missing .functype
    call stop_rewind
    ^
/Users/dkegel/src/tinygo/src/internal/task/task_asyncify_wasm.S:48:5: error: symbol stop_unwind missing .functype
    call stop_unwind
    ^
/Users/dkegel/src/tinygo/src/internal/task/task_asyncify_wasm.S:76:5: error: symbol start_rewind missing .functype
    call start_rewind
    ^
failed to run tool: clang
panic: expected inttoptr

goroutine 30 [running]:
github.com/tinygo-org/tinygo/transform.LowerFuncValues(0x34605100)
	/Users/dkegel/src/tinygo/transform/func-lowering.go:186 +0x1265
github.com/tinygo-org/tinygo/transform.Optimize(0x34605100, 0xc000230540, 0x2, 0x2, 0x5, 0x0, 0x0, 0x0)
	/Users/dkegel/src/tinygo/transform/optimizer.go:83 +0xf77
github.com/tinygo-org/tinygo/builder.optimizeProgram(0x34605100, 0xc000230540, 0xc0047e89d0, 0x2)
	/Users/dkegel/src/tinygo/builder/build.go:855 +0x14f
github.com/tinygo-org/tinygo/builder.Build.func2(0xc0044f88a0, 0x0, 0x0)
	/Users/dkegel/src/tinygo/builder/build.go:473 +0x7e9
github.com/tinygo-org/tinygo/builder.runJob(0xc0044f88a0, 0xc00246a4e0)
	/Users/dkegel/src/tinygo/builder/jobs.go:222 +0xa7
created by github.com/tinygo-org/tinygo/builder.runJobs
	/Users/dkegel/src/tinygo/builder/jobs.go:123 +0x9d8
```

dkegel-fastly, 2022-01-18:
I thought I ran into this, and I did in fact see a very similar panic traceback, but in my case the problem was from accidentally omitting -scheduler=asyncify.

niaow, 2022-01-19:
This has been resolved by #2448 (func lowering no longer exists)

closed 2022-01-26 by deadprogram