tinygo-org/tinygo #2519: wasm: panic: unimplemented: (reflect.Type).Name()

Amari-Mecheri, 2022-01-13:
Hello,

tinygo version 0.21.0 linux/amd64 (using go version go1.17.6 and LLVM version 11.0.0)

Whenever I try to use json.Marshal with a struct I have this error message

panic: unimplemented: (reflect.Type).Name()

example:

	type Size struct {
		Width  int
		Height int
	}
	size := Size{
		Width: 10,
		Height: 10,
	}
	result, _ := json.Marshal(size)
	fmt.Println(string(result))

expected: {"Width":10,"Height":10}
got: panic: unimplemented: (reflect.Type).Name()

dgryski, 2022-01-13:
There's a partial fix here: https://github.com/tinygo-org/tinygo/pull/2479

Full reflect support is tricky.  You might be better off if you can use a code-generation based json package.  People have had more success with those.

Amari-Mecheri, 2022-01-13:
Ok will do that. Thank you

closed 2022-01-13 by Amari-Mecheri