tinygo-org/tinygo #2530: :bug: undefined symbol: os.runtime_args & errno on target "nintendoswitch" when import "fmt"

ca1e, 2022-01-16:
```golang
package main
import "fmt"
func main() {
	fmt.Println("Hello,world")
}
```
$tinygo build -target nintendoswitch -o dummy
```
ld.lld: error: undefined symbol: os.runtime_args
>>> referenced by scheduler_any.go:19
ld.lld: error: undefined symbol: errno
>>> referenced by syscall_libc.go:0
```

ca1e, 2022-01-16:
go version: 1.17.6
tinygo version:  0.21.0

ca1e, 2022-01-16:
tinygo: version 0.22.0-dev-751ac85, here is another errors:
```
# os
~\tinygo\src\os\types_unix.go:20:18: Stat_t not declared by package syscall
~\tinygo\src\os\stat_linux.go:46:32: Timespec not declared by package syscall
~\tinygo\src\os\file_anyos.go:185:16: S_ISUID not declared by package syscall
~\tinygo\src\os\file_anyos.go:188:16: S_ISGID not declared by package syscall
~\tinygo\src\os\file_anyos.go:191:16: S_ISVTX not declared by package syscall
~\tinygo\src\os\stat_linux.go:19:31: S_IFMT not declared by package syscall
~\tinygo\src\os\stat_linux.go:20:15: S_IFBLK not declared by package syscall
~\tinygo\src\os\stat_linux.go:22:15: S_IFCHR not declared by package syscall
~\tinygo\src\os\stat_linux.go:24:15: S_IFDIR not declared by package syscall
~\tinygo\src\os\stat_linux.go:26:15: S_IFIFO not declared by package syscall
~\tinygo\src\os\stat_linux.go:28:15: S_IFLNK not declared by package syscall
~\tinygo\src\os\stat_linux.go:30:15: S_IFREG not declared by package syscall
~\tinygo\src\os\stat_linux.go:32:15: S_IFSOCK not declared by package syscall
~\tinygo\src\os\stat_linux.go:35:25: S_ISGID not declared by package syscall
~\tinygo\src\os\stat_linux.go:38:25: S_ISUID not declared by package syscall
~\tinygo\src\os\stat_linux.go:41:25: S_ISVTX not declared by package syscall
~\tinygo\src\os\stat_linux.go:52:43: Stat_t not declared by package syscall
~\tinygo\src\os\stat_unix.go:22:37: Stat not declared by package syscall
~\tinygo\src\os\stat_unix.go:35:37: Lstat not declared by package syscall
```

ca1e, 2022-01-18:
> What was the last release of tinygo that worked in (if any)?

seems fine with tinygo version 0.17.0 windows/amd64 (using go version go1.16.13 and LLVM version 11.0.0)

tinygo 0.18..0 shows error: undefined symbol: errno

dkegel-fastly, 2022-01-20:
fwiw, can be reproduced with
```
cd tests/tinygotest
tinygo test -c -target=nintendoswitch
```

dkegel-fastly, 2022-01-20:
#2565 includes a regression test for this, but doesn't run it yet from Makefile, as it is still failing.

Ultracoolguy, 2022-11-24:
I could fix the errno issue in 0.18.0 by changing a line in `runtime_nintendo_switch.go` from `go:extern` to `go:export`. However, I think it's unrelated to the undefined syscall symbols in the current release.

deadprogram, 2024-12-02:
Please see #4648 for a fix.

deadprogram, 2024-12-04:
Tagging to close this issue on next release.

closed 2024-12-21 by deadprogram