tinygo-org/tinygo #2532: stm32: timers allocating excess flash and ram

kenbell, 2022-01-16:
To handle timer 'wrap' and 'compare' interrupts, there is a function like this for all STM32:
```go
func (t *TIM) registerOCInterrupt() interrupt.Interrupt {
	switch t {
	case &TIM1:
		return interrupt.New(irq_TIM1_CC, TIM1.handleUPInterrupt)
	case &TIM2:
		return interrupt.New(irq_TIM2, TIM2.handleOCInterrupt)
	case &TIM3:
		return interrupt.New(irq_TIM3, TIM3.handleOCInterrupt)
	case &TIM6:
		return interrupt.New(irq_TIM6, TIM6.handleOCInterrupt)
	case &TIM7:
		return interrupt.New(irq_TIM7, TIM7.handleOCInterrupt)
	case &TIM15:
		return interrupt.New(irq_TIM1_BRK_TIM15, TIM15.handleOCInterrupt)
	case &TIM16:
		return interrupt.New(irq_TIM1_UP_TIM16, TIM16.handleOCInterrupt)
	}

	return interrupt.Interrupt{}
}
```

This seems to cause a lot of flash to be allocated.
```
$ tinygo build -o=blinky.elf -size=full -target nucleo-l432kc ./src/examples/blinky1/
   code  rodata    data     bss |   flash     ram | package
------------------------------- | --------------- | -------
   1328       0      24      89 |    1352     113 | (unknown)
    528       0       0       0 |     528       0 | C compiler-rt
     14       0       0       0 |      14       0 | C picolibc
      0       0       0    4096 |       0    4096 | C stack
    178       0       0       0 |     178       0 | device/arm
    188       0       0       0 |     188       0 | device/stm32
    278      24       0       0 |     302       0 | internal/task
    204       0     868     147 |    1072    1015 | machine
     20       0       0       0 |      20       0 | main
   2322     546       0      12 |    2868      12 | runtime
     10       0       0       0 |      10       0 | runtime/interrupt
    236       0       0       0 |     236       0 | runtime/volatile
------------------------------- | --------------- | -------
   5306     570     892    4344 |    6768    5236 | total
```

If leave just TIM15 (used for clock) in the switch statements in registerOCInterrupt and registerUPInterrupt, the size is:
```
$ tinygo build -o=blinky.elf -size=full -target nucleo-l432kc ./src/examples/blinky1/
   code  rodata    data     bss |   flash     ram | package
------------------------------- | --------------- | -------
   1144       0       4      74 |    1148      78 | (unknown)
    528       0       0       0 |     528       0 | C compiler-rt
     14       0       0       0 |      14       0 | C picolibc
      0       0       0    4096 |       0    4096 | C stack
    180       0       0       0 |     180       0 | device/arm
     40       0       0       0 |      40       0 | device/stm32
    278      24       0       0 |     302       0 | internal/task
    218       0       0     130 |     218     130 | machine
     20       0       0       0 |      20       0 | main
   2300     546       0      12 |    2846      12 | runtime
     10       0       0       0 |      10       0 | runtime/interrupt
    258       0       0       0 |     258       0 | runtime/volatile
------------------------------- | --------------- | -------
   4990     570       4    4312 |    5564    4316 | total
```

The timer interrupt handling is adding approx 20% to flash and ram in this case.

Using objdump, the main difference seems to be the references to the TIMx variables from the registerXXInterrupt functions causing them (and thus the device/stm32 object) to be allocated.

Current:
```
$ objdump -t ./blinky.elf | grep TIM
08000405 l     F .text	00000014 (*machine.TIM).Count
0800042d l     F .text	00000040 (*machine.TIM).Unset
08000419 l     F .text	00000014 (*machine.TIM).channelCCR
08000383 l     F .text	00000080 (*machine.TIM).handleOCInterrupt$bound
0800034f l     F .text	00000034 (*machine.TIM).handleUPInterrupt$bound
200011e0 l     O .data	00000078 machine.TIM1
20001258 l     O .data	00000078 machine.TIM15
200012d0 l     O .data	00000078 machine.TIM16
20001000 l     O .data	00000078 machine.TIM2
20001078 l     O .data	00000078 machine.TIM3
200010f0 l     O .data	00000078 machine.TIM6
20001168 l     O .data	00000078 machine.TIM7
08000219 g     F .text	00000018 TIM1_BRK_TIM15_IRQHandler
08000255 g     F .text	0000000c TIM1_CC_IRQHandler
08000231 g     F .text	00000024 TIM1_UP_TIM16_IRQHandler
08000261 g     F .text	00000018 TIM2_IRQHandler
08000279 g     F .text	00000018 TIM3_IRQHandler
080002c9 g     F .text	00000018 TIM6_DACUNDER_IRQHandler
080002e1 g     F .text	00000018 TIM7_IRQHandler
080014a5  w    F .text	00000006 LPTIM1_IRQHandler
080014a5  w    F .text	00000006 LPTIM2_IRQHandler
080014a5  w    F .text	00000006 TIM1_TRG_COM_IRQHandler
```

With all but TIM15 case statement commented:
```
$ objdump -t ./blinky.elf | grep TIM
08000311 l     F .text	0000000c (*machine.TIM).Count
08000325 l     F .text	00000034 (*machine.TIM).Unset
0800031d l     F .text	00000008 (*machine.TIM).channelCCR
20001098 l     O .bss	00000001 machine.TIM15.4.0
200010a0 l     O .bss	00000001 machine.TIM15.5.0
200010b8 l     O .bss	00000004 machine.TIM15.6.1
200010c0 l     O .bss	00000004 machine.TIM15.7.0.1
08000219 g     F .text	00000068 TIM1_BRK_TIM15_IRQHandler
08001369  w    F .text	00000006 LPTIM1_IRQHandler
08001369  w    F .text	00000006 LPTIM2_IRQHandler
08001369  w    F .text	00000006 TIM1_CC_IRQHandler
08001369  w    F .text	00000006 TIM1_TRG_COM_IRQHandler
08001369  w    F .text	00000006 TIM1_UP_TIM16_IRQHandler
08001369  w    F .text	00000006 TIM2_IRQHandler
08001369  w    F .text	00000006 TIM3_IRQHandler
08001369  w    F .text	00000006 TIM6_DACUNDER_IRQHandler
08001369  w    F .text	00000006 TIM7_IRQHandler
```

This probably warrants some analysis and to either make the optimization 'smarter' to optimize this case or to refactor the code to be easier to optimize.

aykevl, 2022-01-21:
It might be worth investigating whether inlining `registerOCInterrupt` helps. If the compiler knows the `t` value, it can remove all the unused branches and optimize away the unused timers.

kenbell, 2022-01-22:
I've done some experiments...  writing the code like this fixes it...
```
var tims = make([]*TIM, 17)

func (t *TIM) registerUPInterrupt() interrupt.Interrupt {
	var intr interrupt.Interrupt
	state := interrupt.Disable()

	switch t {
	case &TIM1:
		tims[1] = t
		intr = interrupt.New(irq_TIM1_UP_TIM16, func(i interrupt.Interrupt) { tims[1].handleUPInterrupt(i) })
	case &TIM2:
		tims[2] = t
		intr = interrupt.New(irq_TIM2, func(i interrupt.Interrupt) { tims[2].handleUPInterrupt(i) })
	case &TIM3:
		tims[3] = t
		intr = interrupt.New(irq_TIM3, func(i interrupt.Interrupt) { tims[3].handleUPInterrupt(i) })
	case &TIM6:
		tims[6] = t
		intr = interrupt.New(irq_TIM6, func(i interrupt.Interrupt) { tims[6].handleUPInterrupt(i) })
	case &TIM7:
		tims[7] = t
		intr = interrupt.New(irq_TIM7, func(i interrupt.Interrupt) { tims[7].handleUPInterrupt(i) })
	case &TIM15:
		tims[15] = t
		intr = interrupt.New(irq_TIM1_BRK_TIM15, func(i interrupt.Interrupt) { tims[15].handleUPInterrupt(i) })
	case &TIM16:
		tims[16] = t
		intr = interrupt.New(irq_TIM1_UP_TIM16, func(i interrupt.Interrupt) { tims[16].handleUPInterrupt(i) })
	default:
		intr = interrupt.Interrupt{}
	}

	interrupt.Restore(state)

	return intr
}
```

(fixed in the sense a bunch of flash and RAM is saved, although this isn't optimal)

My suspicion is that `interrupt.New` being special is keeping the pointers to the TIM objects alive in the original version even if the unused branches are eliminated.