tinygo-org/tinygo #2539: support  Atmel  at89c52  ？

chen406226, 2022-01-18:
i wanna to build at89c52 with tinygo 
What shall I do?
-target  ???
TKS

deadprogram, 2022-01-18:
Hello @chen406226 that chip does not have an SVD file in the https://github.com/tinygo-org/cmsis-svd repo, nor the upstream repo. That is needed to generate the `device` file.

See https://github.com/tinygo-org/tinygo/wiki/Adding-a-new-board for some details on adding new boards to TinyGo.

chen406226, 2022-01-19:
Thank you. 
I'll try arduino 
like this project

closed 2022-01-19 by chen406226

aykevl, 2022-01-21:
The AT89C52 appears to use the 8051 instruction set. The 8051 instruction set is not supported by LLVM (yet). So right now it is not possible to support this chip, but maybe in the future.

Sorry about that :(