tinygo-org/tinygo #2554: archive/zip test failure on wasi

dgryski, 2022-01-19:
`TestFSModTime` passes when run by itself but fails when run as part of the entire test suite.  My guess is this is memory corruption or other garbage collection issue.

dgryski, 2022-01-20:
When run as part of the full suite, the `OpenReader` calls returns `open testdata/subdir.zip: errno 54`, which I believe is 
```
/**
 * Not a directory or a symbolic link to a directory.
 */
#define __WASI_ERRNO_NOTDIR (UINT16_C(54))
```

This still smells like memory corruption somewhere, given that this doesn't happen when this test is run individually..

dgryski, 2022-01-20:
Ok, so the crash reproduces if I only run `TestFS` and `TestFSModTime`, *but* it passes with no error if I comment out the test case that tries to open `testdata/subdir.zip`.  Hmm...

dgryski, 2022-01-20:
And indeed, running just those two tests with `-gc=leaking` causes it to pass

dgryski, 2022-01-21:
Ok, there's something weird going on with the code generation and lifetimes I think.

If I enable `gcAsserts`, then it fails with `gc: metadata array is too small`, but *not* if I actually print out the values of variables used in that calculation, which leads me to believe something is breaking those values..

dgryski, 2022-01-24:
K, more bizarre testing.  So, if I bump the stack size to 32768, then `-run=TestFS` passes *but* running the entire test suite seems to just hang somewhere.

dgryski, 2022-01-25:
The hanging test appears to be `TestZip64LargeDirectory`.

aykevl, 2022-01-26:
> If I enable `gcAsserts`, then it fails with `gc: metadata array is too small`, but _not_ if I actually print out the values of variables used in that calculation, which leads me to believe something is breaking those values..

This might make sense. The heap starts after the globals, so any global added/removed can change `heapStart` and thus the calculations. And printing values can certainly affect what optimizations can be applied. You could try to print the absolute minimum you need to reproduce the bug, for example `&heapStartSymbol` should be safe to print.

In any case, it looks like you've found an off-by-one.

dgryski, 2022-01-26:
Fix for the off-by-one https://github.com/tinygo-org/tinygo/pull/2588, but this doesn't solve the underlying issue with the slow test (due likely to triggering pathologic gc behaviour).

dgryski, 2022-02-02:
Retitling to be a more accurate version of the bug that was fixed.  `archive/zip` still fails on wasi due to garbage collector issues.

closed 2022-04-29 by deadprogram