tinygo-org/tinygo #2562: Enhancement request: replace Go's deadcode linter

jzabinski-dolios, 2022-01-20:
When I write Go functions in a `main` package and add the `export` keyword, Go's [deadcode linter](https://github.com/remyoudompheng/go-misc/blob/master/deadcode/deadcode.go) warns that the code is unused. Normally this happens when the code is not otherwise used in the package. But exported functions are intentionally unused.

It would be great to have a deadcode linter intended for TinyGo that makes an exception for `export`ed functions.

aykevl, 2022-01-24:
I think this is a bug in the deadcode linter, not in TinyGo. It should recognize `//export` and not treat those functions as unused.

davecheney, 2026-10-28:
Thanks for raising this. As noted above, the warning comes from the external deadcode linter, so support for recognizing `//export` belongs there rather than in a separate TinyGo-specific linter.

Closing as out of scope for this repository. If this still occurs, please report it to the linter’s maintainers with a small example.

closed 2026-10-28 by davecheney