tinygo-org/tinygo #2563: os: problem with Readlink on baremetal

deadprogram, 2022-01-20:
For some reason there is a new failure in the smoke tests for the TinyGo driver's repo, I think is probably in some way due to the implementation in PR https://github.com/tinygo-org/tinygo/pull/2529

The error is:

```
tinygo build -size short -o ./build/test.hex -target=wioterminal ./examples/rtl8720dn/webclient/
go: downloading golang.org/x/net v0.0.0-20210614182718-04defd469f4e
go: downloading golang.org/x/text v0.3.6
# path/filepath
/usr/local/go/src/path/filepath/symlink.go:103:19: Readlink not declared by package os
make: *** [Makefile:208: smoke-test] Error 1
```

https://app.circleci.com/pipelines/github/tinygo-org/drivers/1104/workflows/c928d9b8-975a-4915-bae9-67ce62e940df/jobs/1139?invite=true#step-105-400

dkegel-fastly, 2022-01-20:
Reproducible without drivers repo using by building path/filepath test for bare metal target, I think, e.g.
```
tinygo test -c -target=pybadge path/filepath
# path/filepath
/usr/local/go/src/path/filepath/symlink.go:103:19: Readlink not declared by package os
```

deadprogram, 2022-01-20:
Confirmed on my machine:

```
$ tinygo test -c -target=pybadge path/filepath
# path/filepath
/home/ron/.gvm/gos/go1.17.5/src/path/filepath/symlink.go:103:19: Readlink not declared by package os
FAIL
```

dkegel-fastly, 2022-01-20:
Fix with proper test coming tonight.

dkegel-fastly, 2022-01-21:
https://github.com/tinygo-org/tinygo/pull/2565 should sort this.

deadprogram, 2022-01-21:
Drivers build went green thanks to you @dkegel-fastly so now closing.

closed 2022-01-21 by deadprogram