tinygo-org/tinygo #2572: undefined symbol: fork

fstanis, 2022-01-23:
I'm using TinyGo to build for an ARM-based linux system and the only piece I'm struggling with is running an external command. As `os/exec` is not supported, I tried to use the `fork` and `execve` system call combination instead. However, it seems `fork` for some reason isn't found by the linker:

```go
package main

// #include <unistd.h>
import "C"

func main() {
	C.fork()
}
```

This results in `tinygo:ld.lld: error: undefined symbol: fork`.

Is there any library / flag I could provide to let the linker know about `fork`? Is there any other way to run an external command from a TinyGo-compiled program?

dkegel-fastly, 2022-01-23:
I fear that way lies danger.

But consider the zygote pattern used by Chrome: a simple process written in C++ forks the more complex process, passing it a pipe over which to send it commands, and hangs around waiting for commands to spawn new subprocesses.

You could have a really small C program launch your tinygo app and handle its process spawning chores.  It's not that much code.

davecheney, 2026-10-28:
The reported linker failure no longer reproduces. The original example builds successfully for Linux/ARM, and builds and runs on Linux/AMD64.

I also tested child creation and exit on Linux/AMD64:

```go
package main

/*
#include <unistd.h>
#include <sys/wait.h>
*/
import "C"

func main() {
	pid := C.fork()
	if pid < 0 {
		panic("fork failed")
	}
	if pid == 0 {
		C._exit(0)
	}
	var status C.int
	if C.waitpid(pid, &status, 0) != pid || status != 0 {
		panic("child did not exit successfully")
	}
	println("fork and waitpid succeeded")
}
```

```text
$ tinygo version
tinygo version 0.44.0-dev-1e0bba2f linux/amd64 (using go version go1.27.3-X:nodwarf5 and LLVM version 23.1.4)

$ tinygo build -o fork-test main.go
$ ./fork-test
fork and waitpid succeeded
```

Closing as fixed for the missing `fork` symbol. Remaining limitations in running external commands through `os/exec` are tracked in #5695.

closed 2026-10-28 by davecheney