tinygo-org/tinygo #2580: "tinygo build" aborts on Alpine with "error: could not find build ID"

dkegel-fastly, 2022-01-24:
https://github.com/tinygo-org/tinygo/pull/2366 seems to have introduced a regression.

Discovered by building and using current dev branch inside an Alpine 3.14 container, which worked in 0.21.

tinygo/builder/buildid.go was expecting the gnu build id to be in a section named .note.gnu.build-id,
but on Alpine, it appears to be in a section named .note.go.buildid. 

The obvious change (accepting either) allows tinygo to successfully build programs on alpine, pull request to follow.

dkegel-fastly, 2022-01-24:
https://github.com/golang/go/blob/84eefdc933410907495e42aac872036403851ffa/src/cmd/internal/buildid/buildid.go#L143 suggests that go.buildid is the proper name, and gnu.build-id was working by accident?

There is also gnu.buildid floating around the internet, just to make things confusinger.

deadprogram, 2022-01-25:
Closing since this was correct in-stream for the current release. Thanks!

closed 2022-01-25 by deadprogram