tinygo-org/tinygo #2582: PWM example for Pico won't compile

bmentink, 2022-01-25:
It seems the "machine.NewPWM()" function is not available anymore, does this example need to be updated?

soypat, 2022-01-25:
Yes! There was a change to the TinyGo PWM API a while ago and the examples were not updated. 

Below is what I believe would be proper usage of the API nowadays.
```go
pwm := machine.PWM0
err := pwm.Configure(machine.PWMConfig{})
if err != nil {
    panic(err.Error())
}
ch,err:=pwm.Channel(myPin)
if err != nil {
    panic(err.Error())
}
pwm.Set(ch, pwm.Top()/2)
```

bmentink, 2022-01-25:
Many thanks for that. I did in the meantime find some examples in "/usr/local/lib/tinygo/src/examples" including PWM.

By the way, is their any support for the PIO module(s) on the RaspberryPI pico?, if not can I please request that be put on the TODO list ....

Many Thanks

deadprogram, 2022-05-24:
This has been updated and also the tutorial on the site, so now closing.

closed 2022-05-24 by deadprogram