tinygo-org/tinygo #2587: rs/zerolog thinks context.DeadlineExceeded is nil

dgryski, 2022-01-26:
This code has different behaviour under `tinygo` and `go`.

```
package main

import (
	"unsafe"
)

func isNilValue(i interface{}) bool {
	return (*[2]uintptr)(unsafe.Pointer(&i))[1] == 0
}

type empty struct{}

var errEmpty = empty{}

func (e empty) Error() string {
	return "empty error"
}

func printErr(s string, err error) {
	if isNilValue(err) {
		println(s, "nil value")
	} else {
		println(s, "got an error")
	}
}

func main() {
	printErr("empty", errEmpty)
}
```

aykevl, 2022-01-26:
I feel like I've seen this issue before.
In essence, this package relies on implementation defined behavior and so it is not portable. If it wants to be portable, it shouldn't try to read the interface value like this.

fgsch, 2022-01-26:
@aykevl We talked about this a few months ago :)

dgryski, 2022-01-26:
Hmm.. is there an implementation safe way to handle this? Patching upstream is possible.

dkegel-fastly, 2022-02-19:
Would filing a bug at https://github.com/rs/zerolog/issues make sense, at least?  I didn't see one there yet.

aykevl, 2022-02-19:
This seems to be close to a portable implementation:
https://go.dev/play/p/pH4fFRTRrx-
But of course I don't know what the original intention of the `isNilValue` was.

Also, I somehow managed to find a bug in the Go type checker??
https://go.dev/play/p/aQ237Lsq3K0

davecheney, 2026-10-28:
This is fixed in current zerolog. I tested v1.35.1 with TinyGo `0.44.0-dev-eae3f875` on macOS/ARM64, checking the JSON output from `logger.Error().Err(err).Msg("test")`.

Both `go test -v` and `tinygo test -v` passed these cases:

| Error value | Result |
|---|---|
| `context.DeadlineExceeded` | `"error":"context deadline exceeded"` |
| The report’s empty-struct error | `"error":"empty error"` |
| `nil` | No error field |
| Typed-nil pointer | No error field |

Zerolog now uses reflection for this check rather than inspecting the interface’s memory layout.

Closing as fixed upstream. Thanks for reporting it.

closed 2026-10-28 by davecheney