tinygo-org/tinygo #2591: DotStar support for the itsybitsy m0

Arsen6331, 2022-01-28:
The itsybitsy m0 and some other adafruit boards have an integrated DotStar LED. There is currently no support for it in TinyGo as far as I can tell. 

Here is the arduino library for DotStar LEDs: https://github.com/adafruit/Adafruit_DotStar

deadprogram, 2022-01-28:
The DotStar is actually an APA102. Please see https://github.com/tinygo-org/drivers/tree/release/examples/apa102

Arsen6331, 2022-01-28:
That should probably be documented somewhere, or at least have a pin alias for it similar to `machine.NEOPIXEL`.

deadprogram, 2022-01-28:
Perhaps you would like to submit a PR to the TinyGo site? See https://github.com/tinygo-org/tinygo-site

Arsen6331, 2022-01-28:
> Perhaps you would like to submit a PR to the TinyGo site? See https://github.com/tinygo-org/tinygo-site

Yes, thank you, I can do that. Where would be a good place to document this? Would I put it in the microcontrollers page for all the boards with a dotstar?

deadprogram, 2022-01-28:
> Would I put it in the microcontrollers page for all the boards with a DotStar?

I would suggest add to each, with something in the "Notes" section.

https://github.com/tinygo-org/tinygo-site/blob/release/content/docs/reference/microcontrollers/itsybitsy-m0.md#notes

Arsen6331, 2022-02-01:
Closed since the PR was merged.

closed 2022-02-01 by Arsen6331