tinygo-org/tinygo #2592: PWM only produces max frequency of 1.8khz

bmentink, 2022-01-29:
No matter what value I give "period" in the code below, I can't get a higher PWM on the pin of 1.8Khz.

`package main

import (
	"machine"
)

func main() {
	// Delay a bit on startup to easily catch the first messages.
	//time.Sleep(time.Second * 2)

	pin := machine.GP16
	// This program is specific to the Raspberry Pi Pico.

	pwm := machine.PWM0
	period := 1e9 / 30000

	println("top:", pwm.Top())

	// Configure the PWM with the given period.
	err := pwm.Configure(machine.PWMConfig{
		Period: uint64(period),
	})
	if err != nil {
		println("pwm configure error", err.Error())
	}

	ch, err := pwm.Channel(pin)
	pwm.Set(ch, pwm.Top()/2) // 1/2 duty
	// pwm.SetInverting(ch, true)
	if err != nil {
		println(err.Error())
	}
	for {
		println("top:", pwm.Top())
	}
}`

soypat, 2022-01-29:
What version of tinygo are you running?

soypat, 2022-01-29:
Nevermind, found the bug. Forgot to add a check if `whole` was zero [here](https://github.com/tinygo-org/tinygo/blob/release/src/machine/machine_rp2040_pwm.go#L281) as a result of the division, in which case the `top` calculation fails. In this case `whole` should be set to 1, that should fix it. Will PR a fix

bmentink, 2022-01-29:
I am using version  0.21.0 of tinyGO. 

pwm.Period is 524288 ..

Hope that helps ..

bmentink, 2022-01-30:
Update: Tried V0.22 ... still can't increase my max frequency ... did the fix go into that build?

Also, I noticed that after the upgrade from 0.21 -> 0.22 that I had to run tinygo as root, normal user can no longer run due to permission issues, error below:

`tinygo flash -target=pico main.go 
tinygo:ld.lld: error: cannot open /usr/local/lib/tinygo/pkg/thumbv6m-unknown-unknown-eabi-cortex-m0plus/compiler-rt/lib.a: Permission denied
tinygo:ld.lld: error: cannot open /usr/local/lib/tinygo/pkg/thumbv6m-unknown-unknown-eabi-cortex-m0plus/picolibc/lib.a: Permission denied
failed to run tool: ld.lld
error: failed to link /tmp/tinygo2262361142/main: exit status 1`

soypat, 2022-01-30:
The fix is still no part of TinyGo, it first needs to be merged. If you want the fix today you'd need to clone my TinyGo fork, switch to branch `rp2040-pwmfreqfix` and build TinyGo from source https://tinygo.org/docs/guides/build/

bmentink, 2022-01-30:
No problem, will wait  till it's merged .. any idea's about the permission issues with tinyGo?

dkegel-fastly, 2022-02-05:
The permissions thing is unrelated, possibly a packaging problem.

Please file a new bug with whoever built that tinygo (possibly this project, I dunno), and whether doing

sudo chmod 644 /usr/local/lib/tinygo/pkg/thumbv6m-unknown-unknown-eabi-cortex-m0plus/compiler-rt/lib.a

helps.

bmentink, 2022-02-17:
@soypat 
Could you point me to your above branch with the fix for PWM freq? I can't see your branch listed in the available branches to checkout:

`git checkout rp2040-pwmfreqfix
error: pathspec 'rp2040-pwmfreqfix' did not match any file(s) known to git
`

dkegel-fastly, 2022-02-17:
It's in his fork, https://github.com/soypat/tinygo/tree/rp2040-pwmfreqfix

Try
```
git remote add soypat https://github.com/soypat/tinygo
git fetch soypat
git checkout rp2040-pwmfreqfix
```

bmentink, 2022-02-17:
Cool, building now ..

bmentink, 2022-02-18:
@soypat 
Seemed to build ok and version is:
`tinygo version 0.23.0-dev-e952fc92 linux/amd64 (using go version go1.17.6 and LLVM version 13.0.0)`

However, I get the following when I compile the project:

`./build/tinygo flash -target=pico main.go 
../.cache/tinygo/goroot-647906ad3368d070a4d452794afa8fe948e67ff2737de7db58dc5a365e299fc5/src/machine/machine_rp2040.go:7:2: package device/rp is not in GOROOT (/home/bmentink/.cache/tinygo/goroot-647906ad3368d070a4d452794afa8fe948e67ff2737de7db58dc5a365e299fc5/src/device/rp)
`

Makes no difference deleting the .cache directory, or if I run it as chroot or not ...

bmentink, 2022-02-18:
Next error is:

`./build/tinygo flash -target=pico main.go 
# os/user
../../../usr/local/go/src/os/user/cgo_lookup_unix.go:21:6: not implemented: build constraints in #cgo line
../../../usr/local/go/src/os/user/getgrouplist_unix.go:18:9: error: implicit declaration of function 'getgrouplist' is invalid in C99
`

Is my version of go correct?  `go version go1.17.6 linux/amd64`

bmentink, 2022-02-18:
ahhh, needs to be executed in the project directory ..

All good now, I have a PWM at the programmed 30Khz .. many thanks.

closed 2022-04-29 by deadprogram