tinygo-org/tinygo #2597: fmt.Scanf()  does not work.

bmentink, 2022-01-30:
I get a  runtime "panic: (319:0x20001970)" whenever I try to use Scanf in the form:

fmt.Scanf("%s", &command)

Is it implemented yet?

deadprogram, 2022-01-31:
What target are you compiling for @bmentink and can you please show a bit more of your code?

dgryski, 2022-02-01:
Does the pico have an `os.Stdin`?  `fmt.Scanf` works for me on native but ends up panicing on wasi due to EOF.  I wonder if its a similar issue.

bmentink, 2022-02-01:
@dgryski  Yes, there is an `os.Stdin`

bmentink, 2022-02-02:
@deadprogram 
Do you need any more info to progress this?

deadprogram, 2022-02-03:
> Yes, there is an os.Stdin

On the rp2040? Hmm, I am not too sure that is true. Have to look into that.

bmentink, 2022-02-04:
@deadprogram 
You might be right. I just assumed there was because the TinyGo plugin/Pico target (VS Code ) gave me the popup prompt for os.Stdin when I started typing it ....

bmentink, 2022-02-06:
I am implementing a brushless BLDC motor drive as an excuse to test how feasible TinyGo is with coding for the RPi Pico. It seems I have run into the 1st road block, by not having a serial in. Serial  is essential for me, as I am controlling this drive over a Bluetooth->Serial connection.

@deadprogram  .. any luck with looking into this ?

deadprogram, 2022-02-06:
You do have serial in/out without any problem for example using the UART. What you do not have is any operating system to provide stdin. Hence why you cannot just use `fmt.Scanf()`.

See https://github.com/tinygo-org/tinygo/blob/release/src/examples/echo/echo.go for some ways to read/write to serial.

bmentink, 2022-02-06:
So how come I have a fmt.Printf() but not a fmt.Scanf()?

deadprogram, 2022-02-06:
Output is mapped so that `println()` can be implemented. Note I am not referring to `fmt.Println()` but just the lower level `println()`.

bmentink, 2022-02-06:
`fmt.Printf()` works for me, haven't tried `fmt.Println`

bmentink, 2022-02-07:
@deadprogram 
I tried both `uart.Read()` and `uart.ReadByte()` and both hung waiting for input, but never returned after pressing  some charcters + CR.

bmentink, 2022-02-07:
Upate:  `fmt.Fscanf()` creates a panic as well, I used it in the following way:

func parseSerialCommands() {
	var command, data string
	for {
		in_string := ReadSerialCommand() // Returns a string
		r := strings.NewReader(in_string)
		fmt.Fscanf(r, "%s %s", command, data)
		fmt.Printf("Command=%s Data=%s\r\n", command, data)
	}
}

bmentink, 2022-02-07:
`fmt.Sscanf(in_string, "%s %s", command, data)` produces a panic for me as well.

I would have thought that this should work, as it's not using stdin ...

sago35, 2022-02-08:
@bmentink 
You can use `fmt.Fscanf()` like below.

```go
fmt.Fscanf(uart, "%s", &command)
```

sago35, 2022-02-08:
Stdin, etc. do not work on the baremetal target at this time.
But I feel like it's time to get moving.
I'll work on it later.

https://github.com/tinygo-org/tinygo/blob/3883550c446fc8b14115366284316290f3d3c4b1/src/os/file_other.go#L11-L15

sago35, 2022-02-08:
PR created.
* #2625