tinygo-org/tinygo #2601: Windows release missing `wasm-opt.exe`

Inkeliz, 2022-02-01:
The new version [22](https://github.com/tinygo-org/tinygo/releases/tag/v0.22.0), doesn't include `wasm-opt.exe` for Windows. That was included in the previous version (/bin/wasm-opt.exe).

-------------

That causes the following error:

```
error: could not find wasm-opt, set the WASMOPT environment variable to override
```

stephane-moreau, 2022-02-01:
Just tested the tinygo version 0.22 on my dev environment every thing worked fine after installing binaryen version 105 from https://github.com/WebAssembly/binaryen/

deadprogram, 2022-02-01:
`scoop install binaryen` will install the needed dependencies. Seems like we still need to update the docs with this info!

zzxwill, 2022-10-29:
I fixed it by `sudo npm i wasm-opt -g`.

deadprogram, 2023-05-23:
Closing since that was previously addressed. Thanks everyone!

GageHoweTamu, 2025-05-15:
> I fixed it by `sudo npm i wasm-opt -g`.

3 years later, this was the fix. Thanks!