tinygo-org/tinygo #2616: 'sub' in constant expression is unhandled

QuLogic, 2022-02-06:
Running tests (using several patches from #2515) with Go 1.18 fails with:
```
i64 sub (i64 ptrtoint ([0 x i8]* @"internal/fuzz._ecounters" to i64), i64 ptrtoint ([0 x i8]* @"internal/fuzz._counters" to i64))
panic: unknown constant expr
```

I _think_ this comes from [this bit of code in the fuzzer](https://github.com/golang/go/blob/d588f487703e773ba4a2f0a04f2d4141610bff6b/src/internal/fuzz/counters_supported.go#L20):
```
	size := uintptr(unsafe.Pointer(&_ecounters)) - uintptr(addr)
```

deadprogram, 2022-03-15:
@aykevl can you please take a look?

aykevl, 2022-03-19:
I'm starting to think it may be better to stub out the entire internal/fuzz package because it makes too many assumptions about the compiler and linker.
(If we ever want to implement fuzz testing in TinyGo, we could revisit this decision).

deadprogram, 2022-03-21:
I would agree with that decision for now.

davecheney, 2026-10-26:
@lmaltsis thanks for the repro, this blows up on 0.44-dev. I have a straight forward fix, but as usual when I find a bug in interp i like to pull on that thread a little harder. I should have a fix for the 0.45 timeframe.

davecheney, 2026-10-26:
The crash was caused by an asymmetry between instruction handling and constant-expression handling in interp.

In the instruction path, pointer `Add` was special-cased: interp applied the offset and stored the resulting pointer. There was no matching case for `Sub`, so subtraction fell through to the runtime-emission path ([`interpreter.go`](https://github.com/tinygo-org/tinygo/blob/35f7b28303a12b9917ae34e278f031b01577c659/interp/interpreter.go#L949-L980), [`runAtRuntime`](https://github.com/tinygo-org/tinygo/blob/35f7b28303a12b9917ae34e278f031b01577c659/interp/interpreter.go#L1277-L1280)).

In the reproducer, runtime emission made LLVM fold the subtraction into a constant expression. `rawValue.set` did not handle binary `Add` or `Sub` constant expressions, so it reached the default case and panicked with `unknown constant expr` ([`memory.go`](https://github.com/tinygo-org/tinygo/blob/35f7b28303a12b9917ae34e278f031b01577c659/interp/memory.go#L1124-L1127)).

The `Add` special case hid the same gap for addition. The fix removes that asymmetry: constant `Add` and `Sub` expressions are not evaluated by `rawValue.set`; it returns `false`, allowing interp to leave them for runtime through the existing fallback ([`getValue`](https://github.com/tinygo-org/tinygo/blob/35f7b28303a12b9917ae34e278f031b01577c659/interp/memory.go#L1310-L1314)).

closed 2026-10-30 by davecheney