tinygo-org/tinygo #2623: Seeed Studio XIAO BLE (and BLE Sense)

therebelrobot, 2022-02-07:
Hi there! I'm wanting to add support for the newest XIAO microcontroller from Seeed Studio, the [XIAO BLE](https://wiki.seeedstudio.com/XIAO-BLE-Sense-Getting-Started/). I've worked with the SAMD XIAO prior to this, but just got my hands on a BLE and a BLE Sense, both based on the Nordic nRF52840 MCU, which is covered by TinyGo in other boards. The XIAO BLE Sense is build on the same tech and even has the same documentation in Seeed's website, so I plan on building that support for both.

I'm trying to follow the CONTRIBUTING.md doc as closely as I can, including making this issue, and will for sure follow [the wiki on adding a new board](https://github.com/tinygo-org/tinygo/wiki/Adding-a-new-board). 

Is there documentation handy that shows how to link the local github build to use for development testing? (If it's in the contributing/wiki ignore me ,I'm still making my way through them both)

deadprogram, 2022-02-11:
Hello @therebelrobot 

Please see https://tinygo.org/docs/guides/build/

therebelrobot, 2022-02-13:
Hi @deadprogram! I was able to get building working locally, thank you for that link.

I had an issue with tinygo not being able to find the msd volume when defined in the new device configuration, do you know of any way of scanning a device for available msd volumes? I'm at a loss for figuring out the correct volume name, and the device's forum suggested one that didn't work for finding it.

therebelrobot, 2022-02-13:
For context, here's the info.txt file that shows up when the device is put into bootloader mode, maybe that'll help?

```
UF2 Bootloader 0.6.2-12-g459adc9-dirty lib/nrfx (v2.0.0) lib/tinyusb (0.10.1-293-gaf8e5a90) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: Seeed XIAO nRF52840
Board-ID: Seeed_XIAO_nRF52840_Sense
Date: Nov 30 2021
SoftDevice: S140 7.3.0
```

Telepresent, 2022-02-17:
Hi,
Is Xiao BLE SENSE compatible with Edge Impulse? I would like to use it for sound classification. Any sample project?

Regards

therebelrobot, 2022-03-17:
@deadprogram you would happen to know if there's a contributor who i could pair with to debug this implementation? I keep getting it really close, but it never runs the code. Someone familiar with nordic devices, nrf52840 in particular, would be really helpful to bounce questions off.

aykevl, 2022-03-19:
> I had an issue with tinygo not being able to find the msd volume when defined in the new device configuration

Some notes:
- The MSD volume name is simply the name of the device when you insert the board and double tap to enter bootloader mode.
- You can find the `serial-port` values using `lsusb` on Linux, or something similar on other operating systems.

I hope that helps!

closed 2022-04-29 by deadprogram