tinygo-org/tinygo #2627: 0.22.0 fails to compile in Arch chroot environment

anatol, 2022-02-09:
I am trying to update [Arch Linux package](https://github.com/archlinux/svntogit-community/tree/packages/tinygo/trunk) for `tinygo` to 0.22.0. If I compile the project at my machine it completes successfully. But if I run the same compilation commands at Arch Linux chroot environment it fails with some obscure LLVM error. It is not clear what exactly caused it. The main difference is that chroot environment is more restrictive and contains less pre-installed packages.

Here is the compilation log just in case if you have any ideas what is going on here https://gist.github.com/anatol/82e8758923e2a3df41e4c3c93b1b633a

Here is the most interesting part
```
-- The CXX compiler identification is GNU 11.1.0
free(): invalid pointer
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace, preprocessed source, and associated run script.
Stack dump:
0.	Program arguments: /build/tin-- Detecting C compiler ABI info
ygo/src/tinygo/llvm-build/bin/clang -O2 -g -DNDEBUG --target=wasm32-wasi -fno-trapping-math -Wall -Wextra -Werror -Wno-null-pointer-arithmetic -Wno-unused-parameter -Wno-sign-compare -Wno-unused-variable -Wno-unused-function -Wno-ignored-attributes -Wno-missing-braces -Wno-ignored-pragmas -Wno-PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace, preprocessed source, and associated run script.
Stack dump:
0.	Program arguments: /build/tinygo/src/tinygo/llvm-build/bin/clang -O2 -g -DNDEBUG --target=wasm32-wasi -fno-trapping-math -Wall -Wextra -Werror -Wno-null-pointer-arithmetic -Wno-unused-parameter -Wno-sign-compare -Wno-unused-variable -Wno-unused-function -Wno-ignored-attributes -Wno-missing-braces -Wno-ignored-pragmas -Wno-unused-but-set-variable -Wno-unknown-warning-option -mthread-model single --sysroot=/build/tinygo/src/tinygo/lib/wasi-libc/sysroot -I/build/tinygo/src/tinygo/lib/wasi-libc/libc-bottom-half/headers/private -I/build/tinygo/src/tinygo/lib/wasi-libc/libc-bottom-half/cloudlibc/src/include -I/build/tinygo/src/tinygo/lib/wasi-libc/libc-bottom-half/cloudlibc/src -MD -MP -o /build/tinygo/src/tinygo/lib/wasi-libc/build/libc-bottom-half/cloudlibc/src/libc/unistd/symlinkat.o -c /build/tinygo/src/tinygo/lib/wasi-libc/libc-bottom-half/cloudlibc/src/libc/unistd/symlinkat.c
1.	<eof> parser at end of file
2.	Optimizer
unused-but-setStack dump without symbol names (ensure you have llvm-symbolizer in your PATH or set the environment var `LLVM_SYMBOLIZER_PATH` to point to it):
-variable -Wno-unknown-warning-option -mthread-model single --sysroot=/build/tinygo/src/tinygo/lib/wasi-libc/sysroot -I/build/tinygo/src/tinygo/lib/wasi-libc/libc-bottom-half/headers/private -I/build/tinygo/src/tinygo/lib/wasi-libc/libc-bottom-half/cloudlibc/src/include -I/build/tinygo/src/tinygo/lib/wasi-libc/libc-bottom-half/cloudlibc/src -MD -MP -o /build/tinygo/src/tinygo/lib/wasi-libc/build/libc-bottom-half/cloudlibc/src/libc/unistd/pread.o -c /build/tinygo/src/tinygo/lib/wasi-libc/libc-bottom-half/cloudlibc/src/libc/unistd/pread.c
1.	<eof> parser at end of file
2.	Optimizer
/build/tinygo/src/tinygo/llvm-build/bin/clang(+0x1eb1937)[0x560352084937]
Stack dump without symbol names (ensure you have llvm-symbolizer in your PATH or set the environment var `LLVM_SYMBOLIZER_PATH` to point to it):
/build/tinygo/src/tinygo/llvm-build/bin/clang(_ZN4llvm3sys15CleanupOnSignalEm+0xac)[0x560352060fcc]
/build/tinygo/src/tinygo/llvm-build/bin/clang(+0x1e3c9b9)[0x56035200f9b9]
/usr/lib/libpthread.so.0(+0x13870)[0x7efef046d870]
/build/tinygo/src/tinygo/llvm-build/bin/clang(+0x1eb1937)[0x5613e9b1a937]
/build/tinygo/src/tinygo/llvm-build/bin/clang(+0x12ed0a9)[0x5603514c00a9]
/build/tinygo/src/tinygo/llvm-build/bin/clang(_ZN4llvm3sys15CleanupOnSignalEm+0xac)[0x5613e9af6fcc]
/build/tinygo/src/tinygo/llvm-build/bin/clang(_ZN4llvm25OuterAnalysisManagerProxyINS_15AnalysisManagerINS_6ModuleEJEEENS_8FunctionEJEE6Result10invalidateERS4_RKNS_17PreservedAnalysesERNS1_IS4_JEE11InvalidatorE+0x1006)[0x560351b5f096]
/build/tinygo/src/tinygo/llvm-build/bin/clang(+0x1e3c9b9)[0x5613e9aa59b9
```

dkegel-fastly, 2022-03-22:
I suspect that's a problem on the arch side... I'd suggest trying again with more up to date arch linux and tinygo dev branch.
Also make sure llvm-symbolizer is installed and on PATH before starting the build, for better logging.

closed 2022-03-22 by anatol

aykevl, 2022-03-23:
Ah yes. LTO can be great in many cases but we have a rather complicated build where LTO would likely cause problems.