tinygo-org/tinygo #2636: Unknown import resetTimer - wasi

JeanArhancet, 2022-02-14:
Hello,

I am trying to use tinygo and wasi but I have an error with wasmtime: `1: unknown import: `env::time.resetTimer` has not been defined`

main.go:
```
package main

 import (
	"io/ioutil"
	"log"
	"net/http"
	"fmt"
 )
 
 func main() {
	resp, err := http.Get("https://jsonplaceholder.typicode.com/posts")
	if err != nil {
	   log.Fatalln(err)
	}
	fmt.Println("Response status:", resp.Status)
	body, err := ioutil.ReadAll(resp.Body)
	if err != nil {
	   log.Fatalln(err)
	}

	sb := string(body)
	fmt.Println(sb)
 }
```
The command: `tinygo build -wasm-abi=generic -target=wasi -o main.wasm main.go`

tinygo version:` tinygo version 0.22.0 darwin/amd64 (using go version go1.17.6 and LLVM version 13.0.0)`

dankegel, 2022-02-20:
Dup of https://github.com/tinygo-org/tinygo/issues/1037 ?

codefromthecrypt, 2022-09-07:
this one should be fixed on next release. However, the code won't work in wasi for another reason, which is that it doesn't support network calls needed for http.

deadprogram, 2022-09-30:
Released as part of `v0.26.0`

closed 2022-09-30 by deadprogram