tinygo-org/tinygo #2642: jbub/banking/iban smoke test fails under wasi?

dkegel-fastly, 2022-02-17:
With current dev:

```
$ go run main.go
$ tinygo run main.go
$ tinygo run -target wasi main.go
iban: invalid check digit
$ cat main.go
package main

import "github.com/jbub/banking/iban"

func main() {
	_, err := iban.New("AL50134113214854624345996786")
	if err != nil {
		println(err.Error())
	}
}
```

dgryski, 2022-02-17:
I'll take a look at this tonight.

dgryski, 2022-02-17:
The iban validation code has an integer overflow.  On native, `int` == `int64`, but wasi is a 32-bit platform, so `int` == `int32`.

Upstream can be patched with:

```
 // https://en.wikipedia.org/wiki/International_Bank_Account_Number#Modulo_operation_on_IBAN
 func calculateMod(value string, code string) (int, error) {
-       var total int
+       var total int64
        for _, c := range reformatIban(value, code) {
-               n := codepointToNum(int(c))
+               n := int64(codepointToNum(int(c)))
                if n < 0 || n > 35 {
                        return 0, ErrInvalidIbanModulo
                }
@@ -102,7 +108,7 @@ func calculateMod(value string, code string) (int, error) {
                        total %= modValue
                }
        }
-       return total % modValue, nil
+       return int(total % modValue), nil
 }
```

closed 2022-02-17 by dgryski