tinygo-org/tinygo #2643: reflect-calculated field offset is incorrect

dgryski, 2022-02-18:
It specifically doesn't like the `tm` struct here (which is a shrunken copy of the `time.Time` struct)

```
package main

import (
	"reflect"
	"unsafe"
)

type tm struct {
	ext int64
	loc *tm // just needs to be a pointer here
}

type S struct {
	T tm
	H [][2]string // Input Request Headers
}

func main() {

	a := S{
		H: [][2]string{
			{"foo", "bar"},
			{"baz", "qux"},
		},
	}

	t := reflect.TypeOf(a)

	offs := t.Field(1).Offset
	calcOffs := uintptr(unsafe.Pointer(&a.H)) - uintptr(unsafe.Pointer(&a))

	println(offs, calcOffs)
}
```

dkegel-fastly, 2022-02-18:
... on wasi.

```
$ go run main.go
16 16
$ tinygo run main.go
0x0000000000000010 0x0000000000000010
$ tinygo run -target wasi main.go 
0x0000000c 0x00000010
```

aykevl, 2022-02-18:
I can reproduce this on all 32-bit architectures I've tested.

The core of the issue appears to be that `int64` is aligned to 8 bytes according to LLVM, but to 4 bytes according to Go (see compiler/sizes.go). These must follow whatever LLVM thinks the alignment is (in this case, 8 bytes).

I think this line in particular is wrong, it may need to check the alignment of an actually large value (such as `complex128`):

https://github.com/tinygo-org/tinygo/blob/3883550c446fc8b14115366284316290f3d3c4b1/compiler/compiler.go#L240

...but fixing that probably needs a good amount of testing (also consider float64, complex32, etc).

dgryski, 2022-02-20:
Will this be fixed by the reflect rewrite or does it need its own PR?

aykevl, 2022-02-20:
It's not a bug in the reflect package: the reflect package simply uses the alignment from the compiler. I think this is a bug in the typechecking phase (where `unsafe.Offsetof` is resolved). Specifically, in the line above (https://github.com/tinygo-org/tinygo/issues/2643#issuecomment-1044681181).

dgryski, 2022-02-25:
I think there's more needed than just that one line.  Even if I hard-code `MaxAlign` to `4` I still get the same mismatch.

aykevl, 2022-03-01:
> Even if I hard-code `MaxAlign` to `4` I still get the same mismatch.

That's the point, it must be a higher value. Probably 8.

aykevl, 2022-03-01:
Fix: https://github.com/tinygo-org/tinygo/pull/2658

dgryski, 2022-03-01:
Ah, then I definitely misunderstood the bug and the fix.

closed 2022-04-28 by deadprogram