tinygo-org/tinygo #2648: functions not exported to wasm when using syscall/js

manuschillerdev, 2022-02-23:
As soon as I try to start using the syscall/js in my functions (because I want to pass strings from Javascript to my wasm function), they do not get exported any longer.

### This works fine:
```go
package main

func main() {
	println("Hello World!!!")
}

//export hello
func hello() {
	println("hello world")
}
```

```bash
# compiles without any error
❯ tinygo build -o wasm/out.wasm -target wasm ./wasm/main.go 
```

```html
<!DOCTYPE html>
<html>
  <head>
    <meta charset="utf-8" />
    <title>Go wasm</title>
  </head>

  <body>
    <script src="wasm_exec_tiny.js"></script>
    <script>
      const go = new Go();
      WebAssembly.instantiateStreaming(
        fetch("./out.wasm"),
        go.importObject
      ).then((res) => {
        go.run(res.instance);
        res.instance.exports.hello();
      });
    </script>
  </body>
</html>
```

### using syscall/js
As soon as I compile this, the function `hello` is not available under `exports` in the browser any longer.
```go
package main

import (
	"syscall/js"
)

func main() {
	println("Hello World!!!")
}

//export hello
func hello(this js.Value, args []js.Value) {
	println("hello world")
}
```

Is there any tutorial or reference on how to pass JSON-Objects back and forth between the browser and go functions via wasm?

ppedziwiatr, 2022-03-06:
Hey @manuschillerdev , since I'm having the exact same issue and you've closed the issue - how did you manage to fix this?

akgunberk, 2022-08-30:
hey @manuschillerdev, did you solve this problem?

wieringen, 2023-07-05:
@manuschillerdev, I have the same issue. How did you solve this problem?

magbicaleman, 2023-07-15:
@manuschillerdev did you just throw it away or figure it out?