tinygo-org/tinygo #2649: go-spew tests fail with "panic: (reflect.Value).Interface: unexported"?

dkegel-fastly, 2022-02-23:
Someone wanted to try yenc, so I tried building it.
Its tests plotz[1] on reflect.NewAt.
The call to NewAt comes via an import of an old version of github.com/davecgh/go-spew/spew.

The current version of spew doesn't use NewAt, but its tests still fail like this:
```
$ tinygo test
# github.com/davecgh/go-spew/spew
bypass.go:72:47: reflect.TypeOf((reflect.Value literal)).FieldByName undefined (type reflect.Type has no field or method FieldByName)
bypass.go:106:47: reflect.TypeOf((reflect.Value literal)).FieldByName undefined (type reflect.Type has no field or method FieldByName)
```

Using -tags safe gets further: 
```
$ tinygo test -tags safe
panic: (reflect.Value).Interface: unexported
```

[1] https://www.dictionary.com/e/word-of-the-day/plotz-2020-12-15/

dkegel-fastly, 2022-02-23:
The spew import comes in via testify.

aykevl, 2022-12-12:
I think this was fixed with https://github.com/tinygo-org/tinygo/pull/3111, can you check whether this bug is still present?

dgryski, 2023-02-01:
```
~/go/src/github.com/davecgh/go-spew/spew $ go mod init
go: creating new go.mod: module github.com/davecgh/go-spew/spew
go: to add module requirements and sums:
	go mod tidy
~/go/src/github.com/davecgh/go-spew/spew $ tinygo test
panic: unimplemented: (reflect.Type).FieldByName()
FAIL	github.com/davecgh/go-spew/spew	0.556s
~/go/src/github.com/davecgh/go-spew/spew $ tinygo test -tags=safe
panic: unimplemented: (reflect.Value).UnsafePointer()
FAIL	github.com/davecgh/go-spew/spew	0.474s
```

dgryski, 2023-03-23:
`spew` now works when built with `-tags=safe`.

deadprogram, 2023-06-14:
This is part of the `v0.28` release so now closing this issue. Thanks!

closed 2023-06-14 by deadprogram