tinygo-org/tinygo #2650: testify tests fail with "f.fn.Type().In undefined (type reflect.Type has no field or method In)"

dkegel-fastly, 2022-02-23:
Ooodles of packages use testify.

Testify's "tinygo test" fails at first because of https://github.com/tinygo-org/tinygo/issues/2649
Getting past that with -tags safe yields
```
mock/mock.go:705:28: f.fn.Type().In undefined (type reflect.Type has no field or method In)
mock/mock.go:751:12: fnType.NumIn undefined (type reflect.Type has no field or method NumIn)
mock/mock.go:754:12: fnType.NumOut undefined (type reflect.Type has no field or method NumOut)
mock/mock.go:754:36: fnType.Out undefined (type reflect.Type has no field or method Out)
```

It's not really clear that filing bugs about reflect problems makes much sense - we know reflect needs work - but
as testify is a fairly central package, I figured it was just barely worth filing a bug.

dkegel-fastly, 2022-11-28:
https://pkg.go.dev/github.com/samber/lo looks like a cool generics library, but of course it uses testify for its tests :-(

aykevl, 2022-12-12:
Yeah, `In` is going to take some time (and might result in some feature / code size tradeoffs). It essentially means we'll have to support reflection on functions. But you can try whether stubbing it out will get packages to compile so they become at least usable.

dolmen, 2023-07-04:
It might be useful to adapt `testify` with build tags to isolate parts of its API that require the `reflect` features that tinygo doesn't support.