tinygo-org/tinygo #2651: No access to argv in target wasm?

dkegel-fastly, 2022-02-23:
On slack, Frank Rehwinkel asked (https://gophers.slack.com/archives/CDJD3SUP6/p1645591985252649):

_Hello. I'ld like to pass testing package arguments to node and the wasm_exec.js it runs to be picked up by the tinygo runtime. It seems the plumbing is almost there. ... it seems easy to use js.Global().Get("process").Get("argv") right in the proc.go init() for os.Args. As it is the only target I build, I can change it directly in my copy._ 

(Offhand, seems like the right place for that might be runtime/runtime_wasm_js.go, analogous to the init routine in runtimeruntime_wasm_wasi.go ?)

deadprogram, 2022-04-28:
I think that https://github.com/tinygo-org/tinygo/pull/2780 addresses this request, does it not?

dkegel-fastly, 2022-04-28:
Indeed, fixed.

closed 2022-04-29 by deadprogram

JounQin, 2022-05-09:
Hi, I'm not sure to understand, there is still no `go.argv` in `wasm_exec.js`, so how can I pass `argv` inside?

Sorry, I'm a bit newbie for go/tinygo.