tinygo-org/tinygo #2653: panic when Interrupts enabled  in RP Pico

bmentink, 2022-02-25:
I am getting: `panic: runtime error: nil pointer dereference `  .. whenever I enable interrupts in my test code.

Is there any way to find where the Panic is occurring? Is there a good example for using Interrupts on the Pico?

I am essentially doing the following to use interrupts:

//Interrupt pin: 
comp_A.Configure(machine.PinConfig{Mode: machine.PinInputPullup})

//Set interrupt (presume interrupt is now enabled)
comp_A.SetInterrupt(machine.PinRising, handlerA_rising)

func handlerA_rising(p machine.Pin) {
	comp_A.SetInterrupt(0, nil)) // Disable interrupt  !! not sure if you have to disable interrupt to ensure another one does not happen in this handler ???
 ///.... do stuff
}

Any idea's

sago35, 2022-02-25:
@bmentink 
The following example appears to be working correctly.

https://github.com/tinygo-org/tinygo/blob/release/src/examples/pininterrupt/pininterrupt.go

Note that you need to add the configuration for rp2040.

```go
//go:build rp2040
// +build rp2040

package main

import "machine"

const (
	buttonMode      = machine.PinInputPulldown
	buttonPinChange = machine.PinRising
)
```

bmentink, 2022-02-25:
Yep, that example works. Guess there is something amiss in code that is called by the interrupt ... without further info from the "panic" I am struggling with how to debug this ...

I have plastered error handling all over the place, but none gets hit ... just the panic ..

bmentink, 2022-02-25:
With some more investigation, I am getting: `machine: no channel available for pin interrupt`  error..

I have the following code setting up the interrupts .. pe() is a print error function wrapping the calls.

// These functions setup the comparator interrupts
func (d *Device) bemf_A_rising() {
	pe("setupAR", d.comp_A.SetInterrupt(machine.PinRising, d.handlerA_rising)) // enable this interrupt, disable the rest
}

func (d *Device) bemf_A_falling() {
	pe("setupAF", d.comp_A.SetInterrupt(machine.PinFalling, d.handlerA_falling))
}

func (d *Device) bemf_B_rising() {
	pe("setupBR", d.comp_B.SetInterrupt(machine.PinRising, d.handlerB_rising))
}

func (d *Device) bemf_B_falling() {
	pe("setupBF", d.comp_B.SetInterrupt(machine.PinFalling, d.handlerB_falling))
}

func (d *Device) bemf_C_rising() {
	pe("setupCR", d.comp_C.SetInterrupt(machine.PinRising, d.handlerC_rising))
}

func (d *Device) bemf_C_falling() {
	pe("setupCF", d.comp_C.SetInterrupt(machine.PinFalling, d.handlerC_falling))
}

I get these errors, when they are executed:

setupBF machine: no channel available for pin interrupt
setupAR machine: no channel available for pin interrupt

How am I filling up channels?

sago35, 2022-02-25:
@bmentink 
Please write the entire source code so I can look into it some more.
Please use `Quoting code` to do so.

You may also want to use gist.

https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code

bmentink, 2022-02-26:
@sago35 
Sorry, it's 1000+ lines at the moment and commercially sensitive ..
What else can I post that will help?

All I can say at this point, is that even just calling one of the above functions that sets' the interrupt results in that error ..

bmentink, 2022-02-26:
I think the problem is I can't seem to assign two handlers, one for each edge. I.E

```
func (d *Device) bemf_A_rising() {
	pe("setupAR", d.comp_A.SetInterrupt(machine.PinRising, d.handlerA_rising)) // enable this interrupt, disable the rest
}

func (d *Device) bemf_A_falling() {
	pe("setupAF", d.comp_A.SetInterrupt(machine.PinFalling, d.handlerA_falling))
}

```

Because they refer to the same pin ... I am having issues.

I could have an interrupt handler that responds to both edges, But then how do I tell which edge it is?

sago35, 2022-02-26:
The following requests have been made, but I don't think anyone has worked on them yet.

#2452

bmentink, 2022-02-26:
Hmm, I will try, but very difficult as it is quit complex ... and also, there is a lot of real-time stuff going on, so hard to debug .. especially as the only options at this point is printf's ..

bmentink, 2022-02-28:
@sago35 
I have found the issue. You can't seem to have a `time.Sleep()` call in an interrupt function, it causes the panic.

So my attempt at correcting this is to have the interrupt signal a Goroutine by doing the following:
```
// These are the interrupt handlers
func (d *Device) handlerA(pin machine.Pin) {
	debounce(d.comp_A)
	trigA <- 0
}
```

then the Goroutine:

```
func (d *Device) doAPhase() {
	for { // Forever
		select { // Wait for one of the event below
		case <-trigA:
			//led.Set(true)
			for sector := 0; sector < 6; sector++ {
				for i := 0; i < 32; i++ {
					d.setDutyA(uint32(duty[index[sector][0]][i]) >> 1) // These three phase duty needs to be loaded into the correct duty register
					time.Sleep(time.Microsecond * 2)
				}
			}
			led.Set(true)
		case <-quitA:
			return
		}
	}
}
```
The above `time.Sleep()` causes a panic and the LED is not lit ... if I comment it out the LED is lit and no panic.

So why can't I have sleeps in a Goroutine either ??

bmentink, 2022-03-01:
Is there a HOWTO somewhere on adding a debugger to VSCode for TinyGo on Linux? I used to use OpenOCD for GDB, but I am guessing there has to be some other tool?

aykevl, 2022-03-01:
There is this PR that I still need to take a look at: https://github.com/tinygo-org/vscode-tinygo/pull/4

What I normally use is `tinygo gdb` (which you use just like `tinygo flash` but it'll drop you into a GDB shell). It may need a little bit of configuration depending on your board but many boards with on-board debugger are already supported. This may be helpful: https://tinygo.org/docs/guides/debugging/

bmentink, 2022-03-02:
Thanks, It will be great to see it integrated into VSC ..

bmentink, 2022-03-02:
By the way. I have my application mostly working now.

 I have actually got it going by using a custom ticker(since machine.Tick() is not supported), my ticker sends to a channel continuously at a rate determined by the receiver function. The overhead on this uP is a  round-trip of 2uS. Means I can run motors over 15,000 RPM if I want.

aykevl, 2022-03-02:
> I have actually got it going by using a custom ticker(since machine.Tick() is not supported), my ticker sends to a channel continuously at a rate determined by the receiver function. The overhead on this uP is a round-trip of 2uS. Means I can run motors over 15,000 RPM if I want.

Note that if you do this outside interrupts, there is a risk that something else (an interrupt, the GC, or another goroutine) does something in that time and increases the duration. If you want reliable execution, you need to control the motors entirely from within interrupts.
I don't know the hardware, so I can't tell whether it's critical to always be on time. Just noting it in case :)

bmentink, 2022-03-02:
@aykevl  Thanks. Yep, aware of the variations, but in this application, seems to work well. I can't use Interrupts by themselves to drive the motor, as TinyGo does not allow me to setup both edges at once, So what I have done is run a free running timer as mentioned, and am syncing that periodically with the positive edge of the commutation signal only.  As a side effect it allows free-running startup to happen automatically ..

aykevl, 2022-03-04:
You may be interested in this page: https://tinygo.org/docs/concepts/compiler-internals/heap-allocation/

You can disable the GC entirely using `-gc=leaking`. It will continue to allocate and never de-allocate. This is fine for applications that never allocate memory after initialization, but in that case you would also never see the GC run because the GC only runs when the heap is full.

> Also, since my regular tick method is based on ` time.Sleep()`, and that is interrupt driven, I presume, how does the GC not affect external interrupts, but effects timer based ones?

Interrupts run at a higher priority. They interrupt whatever the chip is doing (including possibly a GC run, which is why it's not allowed to allocate memory inside an interrupt). In contrast, `time.Sleep()` is deeply integrated with the scheduler. Once all goroutines are sleeping, the scheduler will set up a timer and go to sleep. Once enough time has passed, the timer awakes the chip again and the scheduler continues running goroutines. So while the timer is interrupt based, the scheduler still runs at the base (lowest) priority.

bmentink, 2022-03-04:
@aykevl  Many thanks for the explanation.  I tried the `-print-allocs=.` flag, but see very little heap allocations in my code. There is a heap allocation when main() calls my driver main function, but there is no allocations actually in the driver.
So the GC runs I am getting (every 1..2 minutes) must be TinyOS core related .. I will have a play with the `-gc=leaking` flag ..

bmentink, 2022-03-05:
Is there a way to get an interrupt on PWM timer rollover? That would help me achieve what I want ..

deadprogram, 2023-08-18:
Closing since this appears to be resolved. Please reopen if needed. Thanks!

closed 2023-08-18 by deadprogram