tinygo-org/tinygo #2660: Support for reflect package in TinyGo

srikarrao, 2022-03-02:
Is the reflect package implemented in the latest version of Tinygo? I'm trying to decode the JSON in Go WASM and I have been getting the below error that `reflect.Type.NumMethod() is unimplemented`. I found a couple of issues reported earlier, I'm not sure if this is implemented. However, the [TinyGo's supported libs document](https://tinygo.org/docs/reference/lang-support/stdlib/) says it does allow importing `encoding/json` package.

**Previous Issues:**
https://github.com/tinygo-org/tinygo/pull/907
https://github.com/tinygo-org/tinygo/issues/575

**Versions used:**
```
> go version 
	go1.17.6 darwin/amd64
> tinygo version
	tinygo version 0.22.0 darwin/amd64 (using go version go1.17.6 and LLVM version 13.0.0)
> proxy-wasm-go-sdk version
	github.com/tetratelabs/proxy-wasm-go-sdk v0.16.0
```

**Error logs:**
```
proxy_1        | [2022-03-02 02:08:03.668][1][info][wasm] [source/extensions/common/wasm/context.cc:1170] wasm log: panic: unimplemented: (reflect.Type).NumMethod()
proxy_1        | [2022-03-02 02:08:03.668][1][error][wasm] [source/extensions/common/wasm/wasm_vm.cc:38] Function: proxy_on_tick failed: Uncaught RuntimeError: unreachable
proxy_1        | Proxy-Wasm plugin in-VM backtrace:
proxy_1        |   0:  0x170f - runtime._panic
proxy_1        |   1:  0x326d - (reflect.rawType).NumMethod
proxy_1        |   2:  0x6c23a - (*gopkg.in/square/go-jose.v2/json.decodeState).indirect
proxy_1        |   3:  0x696e5 - (*gopkg.in/square/go-jose.v2/json.decodeState).object
proxy_1        |   4:  0x687dc - (*gopkg.in/square/go-jose.v2/json.decodeState).value
proxy_1        |   5:  0x2af8b - gopkg.in/square/go-jose.v2/json.Unmarshal
proxy_1        |   6:  0x65be4 - (*gopkg.in/square/go-jose.v2.JSONWebKey).UnmarshalJSON
proxy_1        |   7:  0x64cef - proxy_on_tick

```

dkegel-fastly, 2022-03-05:
Only a subset of reflect is implemented.    The supported package page is overdue to have a "passes tests" column, see
https://github.com/tinygo-org/tinygo-site/pull/110

Work is ongoing to improve reflect, see https://github.com/tinygo-org/tinygo/pull/2640.

joeblew99, 2022-04-06:
Hey @srikarrao 

I know that https://github.com/buger/jsonparser is used with tinygo where you need to do JSON serialisation. 

You can find the entry point here at https://github.com/genjidb/genji.js/blob/master/scripts/buildgo.sh#L27, which comiles the golang code using tinygo.

johnkeates, 2022-05-17:
I ran into the same issue with `golang-jwt` in a WASI target, it tries to do a `(reflect.rawType).NumMethod` during JSON decoding and that panics because it's simply not there (yet).

dgryski, 2023-03-23:
`encoding/json` "should" be working once my last few reflect fixes are in.  Please let me know if you're still encountering issues.  I will try to test with golang-jwt.

dgryski, 2023-03-23:
The next release of tinygo will support validating jwt:  (I had to disable the http listener example)

```
~/go/src/github.com/golang-jwt/jwt $ tinygo test -tags="purego noasm" -target=wasi
ok  	github.com/golang-jwt/jwt/v5	1.766s
```

ricardbejarano, 2023-04-18:
Thank you so much! When can we expect that release to happen?

deadprogram, 2023-06-14:
This is part of the `v0.28` release so now closing this issue. Thanks!

closed 2023-06-14 by deadprogram

HarikrishnanBalagopal, 2023-10-06:
Is it possible to implement this? https://github.com/tinygo-org/tinygo/blob/731532cd2b6353b60b443343b51296ec0fafae09/src/reflect/type.go#L1012

dkegel-fastly, 2023-10-06:
no necromancy, please :-). Filed https://github.com/tinygo-org/tinygo/issues/3943 for the cobra problem.