tinygo-org/tinygo #2666: regression: "panic: runtime error: out of memory" error on some tests on target cortex-m-qemu

dkegel-fastly, 2022-03-02:
Using the command

    tinygo test -v -target=cortex-m-qemu $foo

the following tests pass on both 0.21 and 0.23-dev:

container/heap
container/list
container/ring
encoding
hash/fnv
internal/itoa
reflect
testing/iotest
unicode/utf16

but the following tests pass on 0.21 and fail on 0.23-dev:

crypto/des
encoding/ascii85
encoding/hex
math/cmplx
net/mail
text/scanner
unicode/utf8

In all cases, the log shows "panic: runtime error: out of memory".

dgryski, 2022-03-02:
The biggest pain I've had with trying to bisect between `release` and `dev` is the llvm version change.  Maybe start with 3a4e0c94e1cbad7227d311e652a0903b26e47273 and narrow down from there?  I can maybe take a look this evening once I get qemu set up.

aykevl, 2022-03-02:
TinyGo still supports LLVM 11, so that shouldn't be a problem. Just compile with `go install -tags=llvm11`.

dgryski, 2022-03-03:
Unsurprisingly,  bisecting gives a536ddcda8025233ad093f5f76d9d728ab67fa3f which brought in the `regexp` package to `testing`.

dkegel-fastly, 2022-03-03:
Bleah.    Dropping support for -run on small platforms doesn't sound great, but neither does switching to some other sort of matching than regexp...

dgryski, 2022-03-03:
We could build a string matcher that just uses `strings.Contains` on "small" platforms.  That way we still have *some* sort of matching behaviour, but it's not a full regexp engine..

dankegel, 2022-03-03:
Probably the best option.

dkegel-fastly, 2022-03-05:
OK, I'll bite.  Who calls MainStart?

aykevl, 2022-03-08:
I already replied on Slack, but for anybody reading this: try `tinygo list -deps -test -json <package>` to get the path to the generated file (the generated main package is the last package in the output).

dankegel, 2022-03-15:
https://github.com/tinygo-org/tinygo/pull/2711 should fix it.

commit c534fa1b (2022-03-15): On baremetal platforms, use simpler test matcher.  Fixes #2666.

Uses same matcher in testdata (because now we can't replace it in testdata).
closed 2022-04-28 by deadprogram