tinygo-org/tinygo #2669: make _cgo_export.h available to Cgo C files

eliasnaur, 2022-03-03:
TinyGo supports calling a Go function from C in the usual Cgo way of `//export`'ing the Go function and adding a C `extern` declaration matching the Go function signature. To minimize risk of function and declaration mismatch, the Go standard implementation generates a `_cgo_export.h` header file containing all `extern` declarations for exported Go functions.

However, TinyGo doesn't seem to offer `_cgo_export.h`, which adds friction to using otherwise portable Cgo packages.

aykevl, 2022-03-03:
Yes, sounds like a feature we should implement eventually.
After a bit of testing I found that `_cgo_export.h` is only present in separate C files, not in the `import "C"` header. Which makes sense: otherwise you're in a chicken-and-egg situation (you only have the types of the exported function after typechecking, and CGo processing happens before typechecking). So basically, to implement this we'd need to add a temporary directory to the C header search path which includes `_cgo_export.h` generated from the `//export` declarations of that package.

haikyuu, 2022-03-12:
Very interesting. Sorry to jump on the issue @eliasnaur but is it possible to call esp32 libraries from c where they are not supported by tinygo? (e.g the Bluetooth module on esp32)?

eliasnaur, 2022-03-14:
I don't know the answer, sorry.