tinygo-org/tinygo #2674: -gc=leaking causes significant code bloat?

dkegel-fastly, 2022-03-03:
A large app, built with default gc, yields a 5.0MB .wasm file.
With -gc=leaking, the .wasm file is 7.3MB (and seems 2x faster).

At @dgryski and @cfallin's suggestion, I added //go:noinline on func alloc() in gc_leaking.go.
This brought the size down to 4.7MB, reducing its size by 30%.
First quick benchmark seem to show It also slowed down slightly, but was still much faster than without -gc=leaking.

aykevl, 2022-03-04:
Just to check: did you compile with `-no-debug`?

> At @dgryski and @cfallin's suggestion, I added //go:noinline on func alloc() in gc_leaking.go.
> This brought the size down to 4.7MB, reducing its size by 30%.

:thinking: I can't think of a reason why that would have such a large impact. I guess you're compiling with `-opt=2` for better speed?

dgryski, 2022-03-04:
> I guess you're compiling with -opt=2 for better speed?

Correct.

dkegel-fastly, 2022-03-04:
No, I did not compile with -no-debug, and doing so cuts binary size in half again.
I'll hold that in reserve unless we get into trouble...

aykevl, 2022-03-04:
> No, I did not compile with -no-debug, and doing so cuts binary size in half again. I'll hold that in reserve unless we get into trouble...

Ah okay. It's only useful to compare binary size when you compile with `-no-debug`. Debug information is large and will grow larger with every attempt to improve it. If you care at all about binary size, you should use `-no-debug`.

dankegel, 2022-03-12:
Turns out it's not just a size thing.

The wasm from our project got too complicated and made a validator take too long.
The fix in https://github.com/dkegel-fastly/tinygo/commit/3e5f60265511a66e3e5185790a306c3ce4d74a4d makes it happy again, but just using -no-debug doesn't.

commit c55c996c (2022-03-17): gc_leaking.go: don't inline alloc.  Fixes #2674.
closed 2022-04-28 by deadprogram