tinygo-org/tinygo #2683: Feature Request: PWM Overfow interrupt

bmentink, 2022-03-06:
Hi,

Requesting that  support for PWM interrupts be added to the `machine_rp2040_pwm.go ` driver, especially handling for the overflow interrupt.

This is required when you want to update PWM duty synchronous with the PWM cycle ..

From the manual:

```
A slice generates an interrupt request each time its counter wraps (or, if CSR_PH_CORRECT is enabled, each time the counter
returns to 0). This sets the flag corresponding to this slice in the raw interrupt status register, INTR. If this slice’s interrupt is
enabled in INTE, then this flag will cause the PWM block’s IRQ to be asserted, and the flag will also appear in the masked
interrupt status register INTS
```

soypat, 2022-03-06:
What would the function/method look like for this? Is there a function you'd like replicated from the C SDK or micropython?

bmentink, 2022-03-06:
I have no idea at this stage, I will try and find out what  Arduino C code implements..

soypat, 2022-03-06:
Alright, I'll see if I begin dipping my toes into writing code again these days.

bmentink, 2022-03-06:
Cool, many thanks ...

soypat, 2022-03-06:
Please take a look at https://github.com/tinygo-org/tinygo/pull/2684. Let me know if it works for you and what else would be required for your use case.

bmentink, 2022-03-06:
Wow that was quick, thanks will try it out ....

soypat, 2022-03-07:
Right, so apparently theres more to it than just setting up the IRQ. I've tried to dive into the pico SDK interrupt code but have been unable to port some parts of it since it's beyond my understanding on how the NVIC works for ARM processors. I'm really just a hobbyist who really wanted to use the pico and had no experience with embedded programming. My progress with the full interrupts implementation is over here https://github.com/soypat/tinygo/tree/rp2040-sync

bmentink, 2022-03-07:
I don't think you are "just" a hobbyist at all, based on all your code efforts.  I am an embedded engineer, but have no experience with TinyGo, but happy to test code to improve TinyGo as I think it is great ..

I am using TinyGo to implement a sensorless BLDC motor controller at the moment and am struggling with real-time support, so any help with getting full interrupt support would be great.  

I also think the Pico is a great embedded platform and when TinyGo gets full support for the PIO unit, it will be a great combo ..

soypat, 2022-03-07:
Hmmmm, I'm gonna throw a little brainstorm your way: what about shorting a pwm signal to a GPIO pin with an interrupt set? Is that too hacky?

bmentink, 2022-03-07:
Yep, had thought of that, is a bit of a hack, and the pwm update won't be
double buffered anymore..