tinygo-org/tinygo #2685: Build Error on Ubuntu

charles-d-burton, 2022-03-06:
Installed using the .deb package for ubuntu.  I'm getting the following permissions error, it's easy enough to fix, just posting this here because I think the .deb needs the permissions updated

```bash
❯ tinygo build -target pico -o test.u2f
tinygo:ld.lld: error: cannot open /usr/local/lib/tinygo/pkg/thumbv6m-unknown-unknown-eabi-cortex-m0plus/compiler-rt/lib.a: Permission denied
tinygo:ld.lld: error: cannot open /usr/local/lib/tinygo/pkg/thumbv6m-unknown-unknown-eabi-cortex-m0plus/picolibc/lib.a: Permission denied
failed to run tool: ld.lld
error: failed to link /tmp/tinygo157798778/main: exit status 1
❯ tinygo version
tinygo version 0.22.0 linux/amd64 (using go version go1.17.7 and LLVM version 13.0.0)
```

charles-d-burton, 2022-03-06:
I fixed it locally with the following command
```bash
❯ sudo find /usr/local/lib/tinygo/pkg/ -type f | sudo xargs chmod 644
❯ tinygo build -target pico -o test.u2f
```

dankegel, 2022-03-08:
Confirming present in 0.21 and 0.22 (but not in 0.20, which didn't contain those files):
```
$ wget https://github.com/tinygo-org/tinygo/releases/download/v0.22.0/tinygo_0.22.0_amd64.deb
$ ar x tinygo_0.22.0_amd64.deb 
$ tar tvf data.tar.gz | "w-----"
-rw------- 0/0          374536 2022-01-25 18:00 ./usr/local/lib/tinygo/pkg/thumbv6m-unknown-unknown-eabi-cortex-m0/compiler-rt/lib.a
-rw------- 0/0          704302 2022-01-25 18:00 ./usr/local/lib/tinygo/pkg/thumbv6m-unknown-unknown-eabi-cortex-m0/picolibc/lib.a
-rw------- 0/0          365756 2022-01-25 18:00 ./usr/local/lib/tinygo/pkg/thumbv7em-unknown-unknown-eabi-cortex-m4/compiler-rt/lib.a
-rw------- 0/0          686310 2022-01-25 18:00 ./usr/local/lib/tinygo/pkg/thumbv7em-unknown-unknown-eabi-cortex-m4/picolibc/lib.a
-rw------- 0/0          375044 2022-01-25 18:00 ./usr/local/lib/tinygo/pkg/thumbv6m-unknown-unknown-eabi-cortex-m0plus/compiler-rt/lib.a
-rw------- 0/0          705066 2022-01-25 18:00 ./usr/local/lib/tinygo/pkg/thumbv6m-unknown-unknown-eabi-cortex-m0plus/picolibc/lib.a
```

dankegel, 2022-03-08:
Fun fact: mac ar cannot deal with those debs because they mistakenly have a slash after the filenames.  Might be an fpm issue...

dankegel, 2022-03-08:
Looks like original sin is committed by tinygo itself?
```
$ umask
0002
$ strace -o slog -e trace=file -f ./build/tinygo build-library -target=cortex-m4     -o build/release/tinygo/pkg/thumbv7em-unknown-unknown-eabi-cortex-m4/picolibc    picolibc
...
88138 mkdirat(AT_FDCWD, "build/release/tinygo/pkg/thumbv7em-unknown-unknown-eabi-cortex-m4/picolibc", 0775) = 0
...
88138 mkdirat(AT_FDCWD, "build/release/tinygo/pkg/thumbv7em-unknown-unknown-eabi-cortex-m4/picolibc/include", 0700) = 0
```
Somehow it's using the wrong permissions when creating that include directory, for instance.

dankegel, 2022-03-08:
Also, this is a bit scary:
```
tar: tinygo: file changed as we read it
```
Perhaps it should not be using "make release deb -j3".

aykevl, 2022-03-12:
So it appears that the underlying cause is that we use `os.TempFile` files which are created with very restricted permissions and we simply copy those files with the same permissions.
I think we should loosen up those permissions a bit.

PR #2691 (2022-03-12): Makefile: chmod before fpm. 
This works around a likely permissions problem in "tinygo build-library"

(Originally removed a -j3, but testing on a hex core machine showed -j3 really does speed things up a bit.)

Fixes #2685
closed 2022-04-29 by deadprogram