tinygo-org/tinygo #2690: abort build if unsupported C function is called

eliasnaur, 2022-03-08:
This program

```
package main

/*
struct c_struct {
	int field1;
	int field2;
};

struct c_struct make_c_struct() {
	struct c_struct s = {
		.field1 = 0xdead,
		.field2 = 0xbeef,
	};
	return s;
}
*/
import "C"
import "fmt"

func main() {
	s := C.make_c_struct()
	fmt.Printf("s.field1 %#x, s.field2 %#x\n", s.field1, s.field2)
}
```

demonstrates that TinyGo doesn't support calling C functions that return structs. On macOS:

```
$ tinygo run structs.go
lld: warning: Option `-demangle' is undocumented. Should lld implement it?
lld: warning: Option `-dynamic' is not yet implemented. Stay tuned...
lld: warning: Option `-install_name' is not yet implemented. Stay tuned...
lld: warning: Option `-dead_strip' is not yet implemented. Stay tuned...
s.field1 0xdead, s.field2 0x345b9850
$ go run structs.go
s.field1 0xdead, s.field2 0xbeef
```

This issue took me a while to find during a port of a Cgo package to TinyGo. Supporting such calls would be ideal, but I'm more than happy if TinyGo would abort the build if incompatible calls are present.

haikyuu, 2022-03-12:
So calling c code that returns an array for example works fine right?

aykevl, 2022-03-12:
@eliasnaur After a bit of playing around it appears that Clang lowers such return values in a different way than TinyGo does (on x86_64, this varies by architecture). Clang converts the struct to a `i64` and returns it that way, while TinyGo returns the (LLVM) struct directly. Most likely, this results in a different effective ABI.

So, it's not really literally _unsupported_, but rather the TinyGo behavior is (sometimes) incompatible with the C ABI.
Not sure how this can best be fixed. We probably need to change the way that parameters and return values are lowered (see compiler/calls.go).

eliasnaur, 2022-03-14:
> @eliasnaur After a bit of playing around it appears that Clang lowers such return values in a different way than TinyGo does (on x86_64, this varies by architecture). Clang converts the struct to a [...]

Just to be clear, I found the issue on my Cortex-M4 board but it was easier to make the report in x86_64 where the issue (appears to be) the same.