tinygo-org/tinygo #2692: don't require -o for tinygo build

eliasnaur, 2022-03-09:
`tinygo build` is useful for checking that a program can build without having to flash it. However, requiring `tinygo build -o /dev/null` or `tinygo build -o program.hex` is cumbersome. `go build` doesn't require an output file, so can `tinygo build` also choose a default output file if `-o` is left unspecified?

dankegel, 2022-03-11:
This would make it easier for people coming from the wider go world.  I saw one user who made the switch complain that tinygo "required extra arguments", and he was probably referring to -o.

aykevl, 2022-03-12:
Yes, this is a good idea. Here is a PR which implements it: https://github.com/tinygo-org/tinygo/pull/2706

deadprogram, 2022-03-14:
Reopening, will close after this is actually released. Hence tagged with that.

closed 2022-04-29 by deadprogram