tinygo-org/tinygo #2693: esp32 in windows10 run fail?

seth-shi, 2022-03-09:
I have installed the esp32 toolchain in MINGW32.
But how should I use TinyGo to run it?

****

I'm running`tinygo flash` on Windows: exec: `xtensa-esp32-elf-ld": executable file not found in %PATH% `
I'm running`tinygo flash` on MINGW32: exec:  `-bash: tinygo: command is not found`

dedalqq, 2022-07-01:
@seth-shi, just install esp-idf on your PC: https://dl.espressif.com/dl/esp-idf/
and then add {esp installed place}\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin in to PATH env for esp32 (use other path for esp32s2, esp32s3 or other)

Gustavomurta, 2023-11-15:
I can't compile TinyGo for ESP32 on Windows 10. I discovered that to run the esptool tool on Linux, **esptool.py** is used. But in Windows 10 there is no such command, after installing esptool. The command is **esptool** (without .py).

`C:\Users\jgust>esptool version`
`esptool.py v4.6.2`

Therefore, when trying to compile TinyGo for ESP32 I receive this error message.

**"esptool.py": executable file not found in %PATH%**

```
PS C:\Users\jgust\tinygo\programas\esp32\flash_spi\flash_example\console\spi> tinygo flash -target=esp32-coreboard-v2 main.go
error: failed to flash C:\Users\jgust\AppData\Local\Temp\tinygo4028449839\main.bin: exec: "esptool.py": executable file not found in %PATH%
```

```
C:\Users\jgust>tinygo env
GOOS="windows"
GOARCH="amd64"
GOROOT="C:\\Program Files\\Go\\"
GOPATH="C:\\Users\\jgust\\go\\"
GOCACHE="C:\\Users\\jgust\\AppData\\Local\\tinygo"
CGO_ENABLED="1"
TINYGOROOT="C:\\Users\\jgust\\scoop\\apps\\tinygo\\current"
```

Can anyone help with this problem? Thank you.

aykevl, 2023-11-18:
@Gustavomurta that's a different issue.

Closing since the original issue was fixed a long time ago in https://github.com/tinygo-org/tinygo/pull/3264.

closed 2023-11-18 by aykevl

Gustavomurta, 2023-11-18:
Sorry by my mistake.

sago35, 2023-11-18:
The issue with esptool has been moved to the following issue.
#3998