tinygo-org/tinygo #2696: SPI.Tx() in atsamd51 is slowing down in tinygo 0.21

sago35, 2022-03-10:
Benchmarks using pyportal_boring show a clear SPI slowdown as of version 0.21.

I put D6.Toggle() in the following location to measure the waveform.
Is this because HasBit() is slow?
Need to look into this a bit more.

```go
func (pd *spiDriver) write8sl(b []byte) {
	machine.D6.Toggle()
	pd.bus.Bus.CTRLB.ClearBits(sam.SERCOM_SPIM_CTRLB_RXEN)

	for i, c := 0, len(b); i < c; i++ {
		for !pd.bus.Bus.INTFLAG.HasBits(sam.SERCOM_SPIM_INTFLAG_DRE) {
		}
		machine.D6.Toggle()
		pd.bus.Bus.DATA.Set(uint32(b[i]))
	}

	pd.bus.Bus.CTRLB.SetBits(sam.SERCOM_SPIM_CTRLB_RXEN)
	for pd.bus.Bus.SYNCBUSY.HasBits(sam.SERCOM_SPIM_SYNCBUSY_CTRLB) {
	}
}
```

tinygo 0.20 : 36 fps
[image]

tinygo 0.21 : 25 fps : There is a large gap between bytes in the SPI transmission.
[image]

source code
https://github.com/tinygo-org/drivers/tree/release/examples/ili9341/pyportal_boing
https://github.com/tinygo-org/drivers/blob/a15f2167cc7e46864d82e08904a3ddaa3e610d5e/ili9341/spi_atsamd51.go#L56-L68

deadprogram, 2022-04-28:
@sago35 is this still an issue with the latest `dev` branch?

deadprogram, 2023-05-15:
Closing since I think this was addressed. @sago35 please reopen if needed.

closed 2023-05-15 by deadprogram