tinygo-org/tinygo #2699: optimize map[string] lookup from []byte key

FrankReh, 2022-03-10:
The current system works but enlists an allocation and memcopy when it calls runtime_stringFromBytes.

The go compiler team did this with their commit:
    https://github.com/golang/go/commit/f5f5a8b6209f84961687d993b93ea0d397f5d5bf#diff-3437cd20ec7506421b8d8b653efa9bfe
    "cmd/gc, runtime: optimize map[string] lookup from []byte key"

Maybe tinygo can do something similar.

If tinygo could enlist escape analysis, it might even go further to use the string([]byte) construct generally for variables or argument passing.

As it stands, there is a mechanism enlisting unsafe used in the wild and the strings package for converting a byte slice to a string without the allocation and copy. And the support by tinygo for unsafe already makes it work. See the example further down.

The basic map lookup under consideration:
```go
func bytes2StringLookup(m map[string]bool, b []byte) bool {

	return m[string(b)]

}
```

This is compiled by tinygo to this wasm function (note the call to runtime_stringFromBytes):

```
function bytes2StringLookup(a:int, b:int):int {
  var c:int = stack_pointer - 32;
  stack_pointer = c;
  c[5]:long@4 = 1L;
  var d:int = 99708[0]:int;
  99708[0]:int = c + 16;
  c[4]:int = d;

  runtime_stringFromBytes(c, b, 32, 32);

  99708[0]:int = d;
  (c + 24)[0]:int = (d = c[0]:int);
  runtime_hashmapStringGet(a, d, c[1]:int, c + 15, 1);
  d = c[15]:ubyte;
  stack_pointer = c + 32;
  return d;
}
```

Here is the unsafe workaround that can be applied manually when a developer wants to take the responsibility for misuse into their own hands. This conversion is already used by the strings.Builder and works in tinygo as tinygo supports strings.Builder without modification.

```go
func unsafeGetString(b []byte) string {
	return *(*string)(unsafe.Pointer(&b)) // As done in strings.Builder
}

func bytes2UnsafeStringLookup(m map[string]bool, b []byte) bool {
	return m[unsafeGetString(b)]
}
```

Some benchmarks showing bytes, strings, and manual/unsafe conversions between the two. Starred are the relevant results for this issue.

```
tinygo/wasm running in nodejs on older mac:
BenchmarkRangeBytes
BenchmarkRangeBytes              	83960235	        14.45 ns/op
BenchmarkRangeBytes              	82679247	        14.36 ns/op
BenchmarkRangeBytes              	83991829	        14.38 ns/op
BenchmarkRangeStringAsBytes
BenchmarkRangeStringAsBytes      	15033096	        79.55 ns/op
BenchmarkRangeStringAsBytes      	14958260	        79.64 ns/op
BenchmarkRangeStringAsBytes      	14833155	        79.80 ns/op
BenchmarkRangeUnsafeStringAsBytes
BenchmarkRangeUnsafeStringAsBytes	83963244	        14.51 ns/op
BenchmarkRangeUnsafeStringAsBytes	81523156	        14.43 ns/op
BenchmarkRangeUnsafeStringAsBytes	83891115	        14.51 ns/op
BenchmarkMapStringLookup
BenchmarkMapStringLookup         	15326641	        78.12 ns/op *
BenchmarkMapStringLookup         	15441502	        79.49 ns/op
BenchmarkMapStringLookup         	15446286	        77.92 ns/op
BenchmarkMapBytesLookup
BenchmarkMapBytesLookup          	 8191792	       144.5 ns/op *
BenchmarkMapBytesLookup          	 8333955	       143.1 ns/op
BenchmarkMapBytesLookup          	 8344280	       142.8 ns/op
BenchmarkMapUnsafeBytesLookup
BenchmarkMapUnsafeBytesLookup    	15252482	        78.89 ns/op *
BenchmarkMapUnsafeBytesLookup    	14966283	        79.39 ns/op
BenchmarkMapUnsafeBytesLookup    	15070021	        79.89 ns/op
```

And the full benchmark code. Works with go, go/wasm and tinygo/wasm.

```go
package main

import (
	"reflect"
	"testing"
	"unsafe"
)

const (
	part = "0123456789"
	s    = part + part + part + "01" // A 32 byte string
)

var save byte

//go:noinline
func rangeBytes(b []byte) {
	for _, b := range b {
		save += b
	}
}

func BenchmarkRangeBytes(b *testing.B) {
	var slice = []byte(s)

	for i := 0; i < b.N; i++ {
		rangeBytes(slice)
	}
}

//go:noinline
func rangeStringAsBytes(s string) {
	for _, b := range []byte(s) {
		save += b
	}
}

func unsafeGetBytes(s string) []byte {
	return (*[0x7fff0000]byte)(unsafe.Pointer(
		(*reflect.StringHeader)(unsafe.Pointer(&s)).Data),
	)[:len(s):len(s)]
}

//go:noinline
func rangeUnsafeStringAsBytes(s string) {
	for _, b := range unsafeGetBytes(s) {
		save += b
	}
}

func BenchmarkRangeStringAsBytes(b *testing.B) {
	for i := 0; i < b.N; i++ {
		rangeStringAsBytes(s)
	}
}

func BenchmarkRangeUnsafeStringAsBytes(b *testing.B) {
	for i := 0; i < b.N; i++ {
		rangeUnsafeStringAsBytes(s)
	}
}

func BenchmarkMapStringLookup(b *testing.B) {
	m := make(map[string]bool)
	m[s] = true
	for i := 0; i < b.N; i++ {
		if !m[s] {
			b.Fatal("failed lookup")
		}
	}
}

//go:noinline
func bytes2StringLookup(m map[string]bool, b []byte) bool {
	return m[string(b)]
}

func BenchmarkMapBytesLookup(b *testing.B) {
	m := make(map[string]bool)
	m[s] = true
	bb := []byte(s)
	for i := 0; i < b.N; i++ {
		if !bytes2StringLookup(m, bb) {
			b.Fatal("failed lookup")
		}
	}
}

func unsafeGetString(b []byte) string {
	return *(*string)(unsafe.Pointer(&b)) // As done in strings.Builder
}

//go:noinline
func bytes2UnsafeStringLookup(m map[string]bool, b []byte) bool {
	return m[unsafeGetString(b)]
}

func BenchmarkMapUnsafeBytesLookup(b *testing.B) {
	m := make(map[string]bool)
	m[s] = true
	bb := []byte(s)
	for i := 0; i < b.N; i++ {
		if !bytes2UnsafeStringLookup(m, bb) {
			b.Fatal("failed lookup")
		}
	}
}
```

dgryski, 2022-03-10:
This seems not too tricky.  I'll take a look at this.

FrankReh, 2022-03-15:
Maybe there should be an optimization pass for bytes to string as there is for string to bytes here:
https://github.com/tinygo-org/tinygo/blob/3883550c446fc8b14115366284316290f3d3c4b1/transform/rtcalls.go#L12-L18

dgryski, 2024-10-21:
I think we can do this by writing an `OptimizeBytesToString` that checks if the result if a conversion is *only* passed to a hashmap lookup function with only instructions that are read-only on the original byte slice in between.  (I think we can do that?)

aykevl, 2024-10-22:
> I think we can do this by writing an `OptimizeBytesToString` that checks if the result if a conversion is _only_ passed to a hashmap lookup function with only instructions that are read-only on the original byte slice in between. (I think we can do that?)

This is more difficult than it may seem. Because Go does not have strict pointer aliasing guarantees (you can cast a pointer to a different type and can often still legally use it) any store to memory (and therefore any function call that might store to memory) would make the optimization impossible.
I tried making a generic `OptimizeBytesToString` pass a few days ago but got stuck on this particular issue.

But yeah, we could do this. I'm not entirely sure the added compiler complexity is worth the optimization benefit though.

closed 2026-11-09 by deadprogram