tinygo-org/tinygo #2702: tinygo wasm externref support

wuhuizuo, 2022-03-11:
Because reflect support issue, such as https://github.com/tinygo-org/tinygo/issues/2660
I try to import host func into wasm module，my steps:

define go import function:
```go
//go:wasm-module json
//export ExternrefTest
func Marshal(data interface{}) ([]byte, error) {
   return []byte{'1'}, nil
}
```

build with wasi target, but got import func type: `func (param i32 i32 i32)`

I import the func with host's golang `encoding/json.Marshal`:
```go
// ...
linker := wasmtime.NewLinker(store.Engine)
linker.DefineFunc(store, "json", "Marshal", json.Marshal)
// ...
```

but i got incompatible import error:
```log
panic: incompatible import type for `json`

Caused by:
    0: instance export "Marshal" incompatible
    1: function types incompatible: expected func of type `(i32, i32, i32) -> ()`, found func of type `(externref) -> (externref, externref)` [recovered]
	panic: incompatible import type for `json`
```

Are there any milestones to support compiled to externref?

codefromthecrypt, 2022-09-07:
I think main issue will be how to represent what should be an externref param unambiguously. Also, per spec they are supposed to be completely opaque from the perspective of the guest. So, if represented by `any`, it should be impossible to cast it to a real type.

aykevl, 2022-09-15:
Right now we don't support externref. Also I don't see how this is going to help you: TinyGo interface values are different from Go interface values. They cannot be interchangeable. If they were, implementing the missing reflect functions would be trivial.

I don't see anything to do here right now.

codefromthecrypt, 2022-09-15:
@aykevl externref is a way to opaquely pass a value the host has in a way that the wasm compiled by tinygo never uses. It is intentional that it isn't interchangeable in other words.

rust doesn't yet support externref either, hence a package like this: https://github.com/slowli/externref

The main thing you get with externref is for the host to be able to stash scoped objects either as globals or stack parameters. The [example above](https://github.com/slowli/externref/blob/main/e2e-tests/src/lib.rs) is a host-defined sender. I've seen other things like host-defined memory as examples (sometimes string literals they don't want to actually wash through wasm).