tinygo-org/tinygo #2704: std net package does not supported in wasm/wasi

Cylkal, 2022-03-11:
# Problem description

code example:
```go
package main

import (
	"net"
	"time"
)

func main() {
	_, _ = net.DialTimeout("tcp", ":80", time.Second)
}
```

compile command:
` tinygo build -o demo.wasm -target=wasi ./test.go`

out:
```
# command-line-arguments
test.go:9:13: DialTimeout not declared by package net
```

# Background
When I want to import package `golang.org/x/crypto/ssh`, trying to connect to ssh server in a wasm module, some err occurs while compile:

```
# golang.org/x/crypto/ssh
../../../../go/pkg/mod/golang.org/x/crypto@v0.0.0-20210813211128-0a44fdfbc16e/ssh/tcpip.go:71:54: TCPAddr not declared by package net
../../../../go/pkg/mod/golang.org/x/crypto@v0.0.0-20210813211128-0a44fdfbc16e/ssh/tcpip.go:104:39: TCPAddr not declared by package net
../../../../go/pkg/mod/golang.org/x/crypto@v0.0.0-20210813211128-0a44fdfbc16e/ssh/tcpip.go:183:51: TCPAddr not declared by package net
../../../../go/pkg/mod/golang.org/x/crypto@v0.0.0-20210813211128-0a44fdfbc16e/ssh/tcpip.go:289:13: TCPAddr not declared by package net
../../../../go/pkg/mod/golang.org/x/crypto@v0.0.0-20210813211128-0a44fdfbc16e/ssh/tcpip.go:389:54: TCPAddr not declared by package net
../../../../go/pkg/mod/golang.org/x/crypto@v0.0.0-20210813211128-0a44fdfbc16e/ssh/client.go:173:19: DialTimeout not declared by package net
../../../../go/pkg/mod/golang.org/x/crypto@v0.0.0-20210813211128-0a44fdfbc16e/ssh/server.go:299:28: TCPAddr not declared by package net
../../../../go/pkg/mod/golang.org/x/crypto@v0.0.0-20210813211128-0a44fdfbc16e/ssh/streamlocal.go:47:28: UnixAddr not declared by package net
../../../../go/pkg/mod/golang.org/x/crypto@v0.0.0-20210813211128-0a44fdfbc16e/ssh/streamlocal.go:85:15: UnixAddr not declared by package net
../../../../go/pkg/mod/golang.org/x/crypto@v0.0.0-20210813211128-0a44fdfbc16e/ssh/streamlocal.go:89:15: UnixAddr not declared by package net
../../../../go/pkg/mod/golang.org/x/crypto@v0.0.0-20210813211128-0a44fdfbc16e/ssh/streamlocal.go:99:30: UnixAddr not declared by package net
../../../../go/pkg/mod/golang.org/x/crypto@v0.0.0-20210813211128-0a44fdfbc16e/ssh/streamlocal.go:112:14: UnixAddr not declared by package net
../../../../go/pkg/mod/golang.org/x/crypto@v0.0.0-20210813211128-0a44fdfbc16e/ssh/tcpip.go:27:21: ResolveTCPAddr not declared by package net
../../../../go/pkg/mod/golang.org/x/crypto@v0.0.0-20210813211128-0a44fdfbc16e/ssh/tcpip.go:191:14: TCPAddr not declared by package net
../../../../go/pkg/mod/golang.org/x/crypto@v0.0.0-20210813211128-0a44fdfbc16e/ssh/tcpip.go:231:17: UnixAddr not declared by package net
../../../../go/pkg/mod/golang.org/x/crypto@v0.0.0-20210813211128-0a44fdfbc16e/ssh/tcpip.go:235:17: UnixAddr not declared by package net
../../../../go/pkg/mod/golang.org/x/crypto@v0.0.0-20210813211128-0a44fdfbc16e/ssh/tcpip.go:355:20: TCPAddr not declared by package net
../../../../go/pkg/mod/golang.org/x/crypto@v0.0.0-20210813211128-0a44fdfbc16e/ssh/tcpip.go:372:16: UnixAddr not declared by package net
../../../../go/pkg/mod/golang.org/x/crypto@v0.0.0-20210813211128-0a44fdfbc16e/ssh/tcpip.go:376:16: UnixAddr not declared by package net
../../../../go/pkg/mod/golang.org/x/crypto@v0.0.0-20210813211128-0a44fdfbc16e/ssh/tcpip.go:391:16: TCPAddr not declared by package net
```

As we can see, `ssh` import `net` package which does not support many methods or types it reqiuired.

Is there any plan to support the official `net` library?

dankegel, 2022-03-11:
WASI does seem to have the beginnings of support for sockets, see
https://radu-matei.com/blog/towards-sockets-networking-wasi/
https://github.com/WebAssembly/WASI/pull/312
https://github.com/bytecodealliance/wasmtime/issues/70
https://github.com/WebAssembly/wasi-sockets
and there has been a tiny bit of discussion about it in tinygo slack,
https://gophers.slack.com/archives/CDJD3SUP6/p1645868342147429
but as yet nobody has wired it up.

sago35, 2022-03-11:
I have not tried this, but the following may be helpful.

> tinynet implements a subnet of the [Go net package](https://golang.org/pkg/net/). Because it is based on [unisockets](https://github.com/alphahorizonio/unisockets), it supports more platforms (WASM/JS, WASM/WASI, TinyGo, Go etc.) than the official net package.

https://github.com/alphahorizonio/tinynet

dipankardas011, 2023-05-30:
same issue here is my error log
```log
# golang.org/x/crypto/ssh
../home/tinygo/go/pkg/mod/golang.org/x/crypto@v0.0.0-20220722155217-630584e8d5aa/ssh/tcpip.go:289:13: undefined: net.TCPAddr
../home/tinygo/go/pkg/mod/golang.org/x/crypto@v0.0.0-20220722155217-630584e8d5aa/ssh/tcpip.go:71:54: undefined: net.TCPAddr
../home/tinygo/go/pkg/mod/golang.org/x/crypto@v0.0.0-20220722155217-630584e8d5aa/ssh/tcpip.go:104:39: undefined: net.TCPAddr
../home/tinygo/go/pkg/mod/golang.org/x/crypto@v0.0.0-20220722155217-630584e8d5aa/ssh/tcpip.go:183:51: undefined: net.TCPAddr
../home/tinygo/go/pkg/mod/golang.org/x/crypto@v0.0.0-20220722155217-630584e8d5aa/ssh/tcpip.go:389:54: undefined: net.TCPAddr
../home/tinygo/go/pkg/mod/golang.org/x/crypto@v0.0.0-20220722155217-630584e8d5aa/ssh/client.go:177:19: undefined: net.DialTimeout
../home/tinygo/go/pkg/mod/golang.org/x/crypto@v0.0.0-20220722155217-630584e8d5aa/ssh/server.go:300:28: undefined: net.TCPAddr
../home/tinygo/go/pkg/mod/golang.org/x/crypto@v0.0.0-20220722155217-630584e8d5aa/ssh/streamlocal.go:47:28: undefined: net.UnixAddr
../home/tinygo/go/pkg/mod/golang.org/x/crypto@v0.0.0-20220722155217-630584e8d5aa/ssh/streamlocal.go:85:15: undefined: net.UnixAddr
../home/tinygo/go/pkg/mod/golang.org/x/crypto@v0.0.0-20220722155217-630584e8d5aa/ssh/streamlocal.go:89:15: undefined: net.UnixAddr
../home/tinygo/go/pkg/mod/golang.org/x/crypto@v0.0.0-20220722155217-630584e8d5aa/ssh/streamlocal.go:99:30: undefined: net.UnixAddr
../home/tinygo/go/pkg/mod/golang.org/x/crypto@v0.0.0-20220722155217-630584e8d5aa/ssh/streamlocal.go:112:14: undefined: net.UnixAddr
../home/tinygo/go/pkg/mod/golang.org/x/crypto@v0.0.0-20220722155217-630584e8d5aa/ssh/tcpip.go:27:21: undefined: net.ResolveTCPAddr
../home/tinygo/go/pkg/mod/golang.org/x/crypto@v0.0.0-20220722155217-630584e8d5aa/ssh/tcpip.go:191:14: undefined: net.TCPAddr
../home/tinygo/go/pkg/mod/golang.org/x/crypto@v0.0.0-20220722155217-630584e8d5aa/ssh/tcpip.go:231:17: undefined: net.UnixAddr
../home/tinygo/go/pkg/mod/golang.org/x/crypto@v0.0.0-20220722155217-630584e8d5aa/ssh/tcpip.go:235:17: undefined: net.UnixAddr
../home/tinygo/go/pkg/mod/golang.org/x/crypto@v0.0.0-20220722155217-630584e8d5aa/ssh/tcpip.go:355:20: undefined: net.TCPAddr
../home/tinygo/go/pkg/mod/golang.org/x/crypto@v0.0.0-20220722155217-630584e8d5aa/ssh/tcpip.go:372:16: undefined: net.UnixAddr
../home/tinygo/go/pkg/mod/golang.org/x/crypto@v0.0.0-20220722155217-630584e8d5aa/ssh/tcpip.go:376:16: undefined: net.UnixAddr
../home/tinygo/go/pkg/mod/golang.org/x/crypto@v0.0.0-20220722155217-630584e8d5aa/ssh/tcpip.go:391:16: undefined: net.TCPAddr
```

dipankardas011, 2023-05-30:
Is it worked on?
if not where should I work to help its development?

dipankardas011, 2023-05-31:
Also one more query does tinygo support cobra cli package? I know its not the right place but whom to ask😕

aykevl, 2023-06-01:
@dipankardas011 you can test so for yourself, it only takes a few seconds ;)