tinygo-org/tinygo #2709: No CI test verifying tinyfs compatibility?

dkegel-fastly, 2022-03-13:
https://github.com/tinygo-org/tinygo/commits/release/src/os/filesystem.go shows the commits that added ReadAt() and Seek() methods to os.FileHandle.
These methods are not present in https://github.com/tinygo-org/tinyfs/blob/release/tinyfs.go, which presumably breaks some uses of tinyfs in tinygo
But CI did not flag those as a problem.  Perhaps we need a tinyfs smoke test in "make smoketest" to verify interoperability.

deadprogram, 2022-03-13:
Solved by this PR: https://github.com/tinygo-org/tinygo/pull/2710

dkegel-fastly, 2022-03-13:
Awesome. 

Those FileHandle changes still aren't flagged as a problem, fwiw.

deadprogram, 2022-03-13:
Yes, that seems like the smoke tests in that repo need more coverage.

closed 2022-04-08 by deadprogram