tinygo-org/tinygo #2720: Can only enable one interrupt handler on rp2040

bmentink, 2022-03-17:
It seems I can only get one GPIO interrupt handler at a time to work.

If I do the following:
```
pinA.SetInterrupt(machine.PinFalling, handlerA)
pinB.SetInterrupt(machine.PinFalling, handlerB)
```

.. then only handlerA works, handlerB does not seem to get called.
If I reverse the two lines above. handlerB now works, and handlerA stops working ..

Cheers,
B

bmentink, 2022-03-22:
Any idea's on this?

bmentink, 2022-03-29:
ok,this is a deal breaker for me. I have had to give up on TinyGo for now and have gone back to my favorite language "Forth".
Within a day, I had my code ported and motor running, AND all interrupts (both edges) active. ...... so good to have an interactive development environment again ... :)

kenbell, 2022-06-26:
@deadprogram this should also be fixed by #2931

closed 2022-07-01 by deadprogram