tinygo-org/tinygo #2726: tinygo crashes in TestFstat with go1.18?

dkegel-fastly, 2022-03-20:
The wasi crash in https://github.com/tinygo-org/tinygo/pull/2725
can be reproduced on that branch and with go1.18 using the command
```
WASMTIME_BACKTRACE_DETAILS=1 build/tinygo test -target wasi -v -run TestFstat  os
```
It can be prevented by live-reverting the following change in go1.18:
```
--- /Users/dkegel/src/go/src/io/fs/fs.go	2022-01-15 18:11:12.000000000 -0800
+++ /usr/local/go/src/io/fs/fs.go	2022-03-15 07:08:05.000000000 -0700
@@ -153,4 +153,4 @@
 	Mode() FileMode     // file mode bits
 	ModTime() time.Time // modification time
 	IsDir() bool        // abbreviation for Mode().IsDir()
-	Sys() interface{}   // underlying data source (can return nil)
+	Sys() any           // underlying data source (can return nil)
```

dkegel-fastly, 2022-03-26:
Slightly simpler reproducer:
```
package main

import "os"

func main() {
        f, err := os.Open("main.go")
        if err != nil {
                println("could not open main.go")
		os.Exit(1)
        }

        info, err := f.Stat()
	if err != nil {
                println("could not stat main.go")
		os.Exit(1)
        }
        println("size of main.go is", info.Size())
}
```

dkegel-fastly, 2022-03-26:
Expanding the import and removing unused code, then manually inlining yielded a longer but maybe more interesting reproducer.
[main.txt](https://github.com/tinygo-org/tinygo/files/8355943/main.txt)

dkegel-fastly, 2022-03-26:
Further minimization yielded a smoking gun:
```
package main

type Stat_t struct {
        UnusedSize      int64
}

type FileInfo interface {
        Size() int64
        Sys() any
}

type fileStat struct {
        size    int64
        sys     Stat_t
}

func (fs *fileStat) Size() int64        { return fs.size }
func (fs *fileStat) Sys() interface{}   { return &fs.sys }   // <- changing return type to any rescues the crash

func Stat(name string) (FileInfo, error) {
        var fs fileStat
        fs.size = 37
        return &fs, nil
}

func main() {
        info, err := Stat("main.go")
        if err != nil {
                println("could not stat main.go")
                return
        }

        println("info.Size() crashes, but if it doesn't, size of main.go is", info.Size())
}
```

Patch coming.

QuLogic, 2022-03-27:
I'm confused; I thought `any` is strictly an alias of `interface{}`. If it causes ABI changes, then it seems like a mysterious breakage just waiting to happen for people using tinygo.

dkegel-fastly, 2022-03-27:
Well then, I guess that's the bug: the tinygo compiler does not treat 'any' as 'identical to interface{} in every way'.

dkegel-fastly, 2022-03-27:
The go compiler does not crash on the minimal reproducer.  Seems like rather a smoking gun...

dkegel-fastly, 2022-03-28:
On the theory that some magic 'any' fix comes in via golang.org/x/tools, I tried 
```
$ go version
go version go1.18 darwin/amd64
$ git checkout dkegel-go118
$ go get golang.org/x/tools
$ go mod tidy
$ make test
$ make
$ build/tinygo run ~/tinybug/main.go
panic: runtime error: nil pointer dereference
$ go run ~/tinybug/main.go
info.Size() crashes, but if it doesn't, size of main.go is 37
```
but alas, the small reproducer still seems to reproduce the problem with tinygo.

QuLogic, 2022-03-28:
If you run with `-dumpssa` or `-printir`, you can still see mentions of `any`. But I'm no expert at reading those, so I have no idea if `any` should still be there, or should instead be eliminated by some optimization pass later.

dgryski, 2022-04-01:
Here is an incorrect patch that fixes the issue.  However, it does point to what the problem is and suggest what the proper fix might be.

```
index d2d4b4f4..b2f80cbb 100644
--- a/transform/interface-lowering.go
+++ b/transform/interface-lowering.go
@@ -342,6 +342,7 @@ func (p *lowerInterfacesPass) addInterface(methodsString string) {
 // getSignature returns a new *signatureInfo, creating it if it doesn't already
 // exist.
 func (p *lowerInterfacesPass) getSignature(name string) *signatureInfo {
+       name = strings.ReplaceAll(name, "interface{}", "any")
        if _, ok := p.signatures[name]; !ok {
                p.signatures[name] = &signatureInfo{
                        name: name,
```

dgryski, 2022-04-01:
A better fix:
```
diff --git a/compiler/interface.go b/compiler/interface.go
index c8b76039..3727fc25 100644
--- a/compiler/interface.go
+++ b/compiler/interface.go
@@ -591,23 +591,31 @@ func signature(sig *types.Signature) string {
                        if i > 0 {
                                s += ", "
                        }
-                       s += sig.Params().At(i).Type().String()
+                       s += emptyInterfaceToAny(sig.Params().At(i).Type().String())
                }
                s += ")"
        }
        if sig.Results().Len() == 0 {
                // keep as-is
        } else if sig.Results().Len() == 1 {
-               s += " " + sig.Results().At(0).Type().String()
+               s += " " + emptyInterfaceToAny(sig.Results().At(0).Type().String())
        } else {
                s += " ("
                for i := 0; i < sig.Results().Len(); i++ {
                        if i > 0 {
                                s += ", "
                        }
-                       s += sig.Results().At(i).Type().String()
+                       s += emptyInterfaceToAny(sig.Results().At(i).Type().String())
                }
                s += ")"
        }
        return s
 }
+
+func emptyInterfaceToAny(typ string) string {
+       if typ == "interface{}" {
+               return "any"
+       }
+
+       return typ
+}
```

QuLogic, 2022-04-02:
Interestingly, if you take the test case and add `type any = interface{}` at the top, it doesn't crash when compiled with Go 1.17. Is that because `any` is defined in the same file and not `builtin`, or is it a 1.17/1.18 difference?

if you do `s/any/something/g` and add `type something = interface{}` to the example, will it also crash compiled with Go 1.18?

aykevl, 2022-04-06:
Interesting. I seem to remember a similar issue with `byte` vs `uint8`, and `rune` vs `int32`. I think we should stop using the `.String()` method and serialize the types in a proper way.

aykevl, 2022-04-06:
The fix by @dgryski is probably not correct either, because it wouldn't handle `struct{X any}` correctly.

dgryski, 2022-04-06:
@aykevl Yeah, I figured my patch was hacky but (as I said) it at least pointed to what the issue was and where.

aykevl, 2022-04-06:
Yup, I've also reproduced this using `byte` vs `uint8` in existing TinyGo.

aykevl, 2022-04-06:
@dgryski yes, it's very helpful to have a hacky-but-working fix. It shows where the problem should be fixed :)

aykevl, 2022-04-06:
Short reproducer:

```go
package main

type Foo int

func (f Foo) Byte() byte { return byte(f) }

type Byter interface {
	Byte() uint8
}

func main() {
	var v Byter = Foo(3)
	print("Byte(): ")
	println(v.Byte())
}
```

aykevl, 2022-04-06:
Working on a fix...

dkegel-fastly, 2022-04-07:
Seems fixed.  Leaving open in case you want to close it as part of the release process...?

closed 2022-04-29 by deadprogram