tinygo-org/tinygo #2732: fmt package unusable, arduino target, 0.22.0

stevegt, 2022-03-22:
Getting "error: interp: ptrtoint integer size does not equal pointer size" from fmt.Sprintf, Println, Printf, etc.  Possibly related to #2389.

```
package main

import (
        "fmt"
        "time"
)

func main() {
        for {
                a := fmt.Sprintf("hello")
                _ = a
                time.Sleep(1 * time.Second)
        }
}
```

tinygo version 0.22.0 linux/amd64 (using go version go1.16.3 and LLVM version 13.0.0)

dankegel, 2022-03-22:
Yeah, it's almost certainly a dup of https://github.com/tinygo-org/tinygo/issues/2389

stevegt, 2022-03-22:
I'd recommend keeping this issue open until fmt works again in a released version, even if it does turn out to be a dup of #2389 -- new folks (and potential contributors) need to be able to figure this out quickly; workarounds can be posted here.  

The 0.22.0 println() function also does not work on the Uno -- I can't tell if this, too, is a known bug.  It seems to be causing timing problems or crashes which I haven't tracked down, so I finally wound up using machine.Serial.Write() to get any USB output at all. 

I'm also an open-source maintainer, so know far too well how this feels.  In the spirit of constructive feedback, this was to be my first use of tinygo for a project, and I was looking forward to it, but lost an evening still trying to get things to work even after finding #2389.  Took me a while to figure out that I was getting that compile-time error because I was importing a third-party sensor library that uses fmt.Sprintf internally pretty extensively.  I wasn't importing fmt myself.

Since most articles and blog posts about tinygo assume that either fmt.Println() or println() just work, and https://tinygo.org/docs/reference/lang-support/stdlib/ also confirms that fmt should work, that was a bizarre several hours. 

For this project at least, all of this added together means I'm having to switch to FreeRTOS, which is of course going to distract me from being able to help with #2389, so vicious circle.  I'd say something like "you really need to prioritize fixing fmt so potential contributors don't run away screaming at first contact", but again, I know how this feels.  

Let me know if there's anything I can do to help, even if it's just being a testing monkey.  I want tinygo to work.

dkegel-fastly, 2022-03-22:
Making tests fail until this is fixed seems a bit much, but I did tag this with the 0.23 milestone to give it more visibility.

aykevl, 2022-03-23:
The fmt package mostly works. The problem is not fmt, but the AVR backend (used for Arduino) that is unstable (but slowly improving). If you use a board with a Cortex-M chip (micro:bit, newer "Zero", "M0" or "M4" brander boards) then it should work.

dankegel, 2022-03-26:
For the record, there is a minimal reporoducer at https://github.com/tinygo-org/tinygo/issues/2389#issuecomment-1075331850

dkegel-fastly, 2022-04-07:
Closing as dup of #2389

dan-dekel, 2023-01-09:
getting the same error when running `tinygo test -target=arduino-nano`

aykevl, 2023-01-09:
Duplicate of #2389 so closing.

closed 2023-01-09 by aykevl